snaptec/openWB #6: SoC-Auslesung für BMW i3

iot4all, 2018-04-30:
Implementierung der SoC-Auslesung (BMW i3) über die API des "connected drive"-Prortals mit den Hinweisen dieses links:
https://github.com/jupe76/bmwcdapi

snaptec, 2018-07-30:
Update für alle:
Beta Version läuft bei iot4all

wenn das zuverlässig funktioniert -> in den nächsten releases dabei

iot4all, 2018-07-31:
Du Kevin,

ich teste gerade Nachtladen mit 70% SoC. Im Fenster "Nacht SOC Freitag bis
Sonntag" bei LP1 wird der Vorgabewert nicht gespeichert. Ist nach SAVE bei
Kontrolle immer wieder leer. Kannst Du das beim nächten update mit fixen?
Bei "Nacht SOC Sonntag bis Donnerstag" klappt es im LP1

Bei LP2 klappen die beiden SoC, dafür wird "Nachtladen Uhrzeit ab" nicht
übernommen.

Ansonsten sind die Optionen bei Nachtladen gelinde gesagt VORBILDLICH :-).
Wenn wird nur die SoC-Vorgabe noch bei Sofort Laden reinbekommen, wie
besprochen, wäre das 1A.

Zum Glück ist erst Dienstag und der erste SoC-Wert wird bei LP1
gespeichert. Mal sehen, ob der i3 morgen 70% hat.

In den Stop-button würde ich noch ein Sternchen hinzufügen und darunter
*...ggf. Nachtladen
schreiben.

Viele Grüße
Frank

iot4all, 2018-07-31:
Und am LP2 wird zwar die Nachtladestromstärke auf der Config eingestellt
und gespeichert, aber nicht auf die Hauptseite übernommen. Dort stehen bei
LP2 immer 12A fix.

Good night

iot4all, 2018-07-31:
Ok - er macht erstmal seine Nachtladung.

Nun ist das einzig unschöne, dass man nicht mehr von der Hauptseite aus
Stoppen kann.

Idee:
Den Stop-button teilen in
"Stop+ggf. NL" sowie "Notstop"
oder
"Nachtladen" sowie "Stop"?

Hast Du ev. eine bessere Idee, um manuell auch bei NL stoppen zu können?

VG

snaptec, 2018-08-16:
Ist implementiert

closed 2018-08-16 by snaptec