snaptec/openWB #7: Softwaremodul für AE-PV-Modulinverter (RS 485, Modbus/RTU)

iot4all, 2018-04-30:
Für die häufig bei Kleinanlagen oder reinen Eigenverbrauchsanlagen verwendeten AE-Wechselrichter
http://www.aeconversion.de/de/index.html
wäre die Einbindung in openWB per Software-Modul wünschenswert, da diese alle in DE geforderten Sicherheitsfeatures implementiert haben.

Diese Modul-WR (bis zu 2x 250W) werden mit Modbus/RTU ausgelesen/angesteuert und verwenden RS485-Hardware. Es gibt 2 Varianten:
a) Entweder haben die WR eine eigene RS485-Schnittstelle oder
b) mehrere WR kommunizieren über die Stromleitung per PLC. Zur Datenfilterung wird dann dieses PLC-Gateway
http://www.aeconversion.de/de/plc-gateway.html
genutzt, welches alle Modbusdaten zur Verfügung stellt.
Die Hardwareanbindung erfolgt also ähnlich der SDMxxx-Zähler bzw. der "EVSE DIN mit RS485". 

Es böte sich an, dieses vorh. Stück Software namens **AECLogger** zu nutzen:
Hierfür wurde das Protokoll reengineered, da es nicht offiziell vom Hersteller freigegeben wurde. Aus den ertmittelten Transferdaten wurde ein in C geschriebener Datenlogger mit Leistungssteuerungsfunktion entwickelt, welcher autark funktioniert (alles inkl. websites in C !!!).
https://github.com/akrypth/aeclogger/wiki
Inwieweit eine API möglich wäre, muss noch recherchiert werden.

snaptec, 2018-05-23:
Gibt es irgendwo Details dazu welche Register genutzt werden?
Das würde ich als PV Modul realisieren und am liebsten direkt per Modbus abfragen.

iot4all, 2018-05-23:
Ja - es gibt ein reengineertes Protokoll. Kommt per mail.

closed 2021-02-13 by benderl