snaptec/openWB #9: Anzeige und ggf. Beschreiben der EVSE-Modbusregister

iot4all, 2018-05-16:
Für die angeschlossenen "EVSE WB" / "EVSE DIN" wäre für Entwickler die Anzeige und das manuelle Schreiben der Modbusregister hilfreich, da dies momentan (16.05.2018) nur über die UART-Verbindung zur EVSE möglich ist (per BT-Modul oder USB/UART).
Bei Nutzung in einem RS485-Produktivsystem wird die UART-Schnittstelle durch den RS485-IC belegt, so dass nur ein aufwendiger Umbau den Zugriff ermöglichen würde.
Der Zugriff direkt über openWB erspart dieses procedere. 

Zum Anzeigen der Modbuswerte könnte z.B. ein neuer Button auf der Einstellseite von openWB realisiert werden, der einmalig die Werte anzeigt (unter Angabe der EVDE-Modbus-ID, da mehrere EVSE im System sein könnten).
Das Setzen eines Wertes müsste dann ggf. in einem Untermenü stattfinden.
So wie in der EVSE-APP wäre es ausreichend.

**Aktuelle register EVSE-FW 8 (Stand 16.05.2018)**
**RAM**
1000: Configured Current
1001: Actual Current
1002: Vehicle Status
1003: PP-Limit
1004: Turn off bit
1005: Firmware Version
1006: EVSE Status

**EEPROM**
2000: Current after boot
2001: Modbus Status
2002: Min. Current Value
2003: Analog Input Config
2004: Amps after boot (evse-button)
2005: activate/deactivate EVSE
2006: Current Sharing Mode
2007: PP-Detection

snaptec, 2018-05-20:
Wäre für mich etwas für die „experten status“ Seite, einverstanden?

iot4all, 2018-05-20:
Klar. Irgendwie müssen wir die Register mal angehen - bis dato wurden die
bei allem außer openWB ja kaum genutzt.
VG

snaptec, 2018-05-23:
Das kommt leider erst zusammen mit dem Modbus handler.
Das zyklische auslesen ist unnötig.
Das manuelle macht Sinn, aktuell aber nicht zuverlässig umsetzbar.