snaptec/openWB #16: Integration ModBus Meter MPM3PM

gb1973, 2018-07-30:
Wer hat Interesse den ModBus Zähler MPM3PM von Bernecker Austria zu integrieren.
Der Zähler ist von uns entwickelt (ich bin ehemaliger Fronius Entwickler) und hat idente Register wie ABB Zähler und wird bereits bei Loxone und und Siemens Logo verwendet.
Der Zähler funktioniert natürlich auch über ModBus TCP <-> RTU koppler.
Man müsste nur die Registerdefinition von MP630 anpassen.
[MPM3PModBus.pdf](https://github.com/snaptec/openWB/files/2240380/MPM3PModBus.pdf)

snaptec, 2018-07-30:
Zähler sehen gut aus!
Kann ich gerne implementieren. 
Dafür bräuchte am besten ein Demo Exemplar hier.

gb1973, 2018-07-30:
Das sollte kein Problem sein, kann ich gerne Muster senden :)

snaptec, 2018-07-30:
Sehr schön.
Rest am besten per mail:
info@snaptec.org

gb1973, 2018-07-30:
(mail)

snaptec, 2018-08-08:
als beta für Ladepunkt 1 integriert.
Bitte testen und Feedback geben. 
Wird dann für die anderen Ladepunkte sowie EVU nachtgetragen.

snaptec, 2018-08-27:
Update:
Warte hier noch auf Feedback ...

snaptec, 2018-10-08:
eingebaut mit v0.93

closed 2018-10-08 by snaptec