snaptec/openWB #18: FeatureRequest: Modul für Solarworld EManager / Kiwigrid

philgrill, 2018-08-25:
Hallo snaptec,

ich bekomme über einen SolarWorld E-Manager / Kiwigrid alle 5 sec. in meinem lokalen Netzwerk unter http://192.168.0.185/rest/solarworld/lpvm/powerAndBatteryData folgenden Text ausgegeben:

{"PowerInstalledPeak":4655,"PowerProduced":632,"PowerOut":97.08172362555717,"PowerSelfSupplied":534.9182763744428,"PowerSelfConsumed":534.9182763744428,"PowerConsumed":534.9182763744428,"PowerIn":0,"PowerConsumptionMax":{"2018-08-21":5142.857142857143,"2018-08-19":7200,"2018-08-22":5142.857142857143,"2018-08-18":10909.090909090908,"2018-08-25":3228.6995515695066,"2018-08-24":5070.422535211268,"2018-08-20":6260.869565217391}}

Mit den Werten kann ich die openWB betreiben ohne weitere Zähler zu verbauen.

Ich habe einen Job laufen, der mit die Werte einzeln ausliest, auf eine http Seite schreibt und dann lese ich es mit dem http Modul aus. Das ganze funktioniert, läuft aber nicht sehr stabil.

Ich bekomme das leider nicht hin es als Modul zubauen. Wäre klasse wenn du oder andere User unterstützen könnten.

Ansonsten ein Wahnsinnsprojekt was ihr auf die Beine gestellt habt. Vielen Dank dafür.

Grüße, Philipp

snaptec, 2018-08-25:
Hallo Philipp,

da sich sehr viele Wechselrichter über Json auslesen lassen steht ein Json Modul bereits auf der Liste / ist in to do.
Hier kannst du dann URL sowie den passenden jq Befehl eingeben.
Im gleichen Zug wird der "Hausverbrauch" einstellbar wenn kein EVU Modul verfügbar ist. 
Der ist aktuell noch fix auf 300Watt eingestellt.

philgrill, 2018-08-26:
Hi snaptec, 

danke für den Hinweis mit dem Json Modul. Habe mir dein Fronius WR Modul umgebaut und es funktioniert.

Grüße, Philipp

snaptec, 2018-08-27:
Hi Philipp,
Bitte mal updaten, in der Version 0.9 ist das Json Modul vorhanden.
Das Problem wenn du ein Modul umbaust und es in der Ordnerstruktur unterhalb von /openWB/.. ist, ist das es bei einem Update ueberschrieben wird.
Klappt es mit dem Json Modul?

closed 2018-10-08 by snaptec