snaptec/openWB #20: Keine DC-Fehlerstromdetektion

bodems, 2018-09-27:
Laut Vorschriften muss eine Ladeanlage auch gegen DC-Fehlerströme abgesichert werden. Ein in Haushalten üblicher RCD Typ A macht das nicht (die erkennen nur AC-Fehlerströme), es braucht also einen teuren RCD Typ B (so ca. 270€). Ich konnte dazu keinen Hinweis finden. Was bei kommerziellen Anlagen auch häufig gemacht wird, ist einen separaten DC-Fehlerstromschutz in die Anlage einzubauen, Bender bietet sowas z.B. für die Hutschiene [1] oder als Platine [2] an.

[1] https://www.bender.de/produkte/differenzstrom-ueberwachung/linetraxx_rcmb42
[2] https://www.bender.de/produkte/differenzstrom-ueberwachung/rcmb104

snaptec, 2018-09-27:
Das ist ein schwieriges Thema.
Derzeit muss das erstmal der Elektriker vor Ort entscheiden der die Installation vornimmt.

Eine Zoe braucht für die 3-phasige Ladung meines Wissens definitiv einen FI Typ B.
Der dort mitgelieferte 1-phasige ICCB nicht!?
Andere Hersteller die 1-phasig mit galvanisch getrenntem Board-Lader funktionieren nicht ebenfalls nicht.

Kann jemand belastbare Quellen hierzu nennen?

Ich würde den Hinweis in der Readme hinzufügen das ggf. ein FI Typ B erforderlich ist, einverstanden?

Lizzard44, 2018-09-27:
NUR beim Dreiphasigen laden wird ein FI-TypA-EV benötigt.

Wenn man davon ausgehen kann das ein FI-TypA schon verbaut ist wäre die Empfehlung für einen FI-Typ-B kontraproduktiv.  Der FI-TypB darf NUR vorgeschaltet werden wenn sonst kein FI vorgeschaltet ist.  

Ein FI-TypA-EV würde immer passen - ist aber teuer. :(

[https://www.goingelectric.de/wiki/Normgerechte_Errichtung_von_Ladeinfrastruktur#Gleichfehlerstrom-Schutzeinrichtungen_nachr.C3.BCsten](url)

snaptec, 2018-09-28:
Kurze Info in der Readme eingefügt.
Die reine DC Überwachung in der WB wird durch @iot4all aktuell geplant um hier in der Zukunft eine Möglichkeit zu geben.

snaptec, 2019-05-31:
gibt es...

closed 2019-05-31 by snaptec