snaptec/openWB #21: Keine Auslesung von PP CP?

bodems, 2018-09-27:
Sehe ich das richtig, dass weder bei der Bauvariante mit festem Kabel (wo dann der PP-Widerstand fest angeschlossen ist) noch bei der Bauvariante mit Dose der PP und CP ausgelesen wird? Wie wird dann
1. erkannt, ob ein Fahrzeug angeschlossen ist?
2. gewährleistet, dass bei der Bauvariante mit Dose das Ladekabel nicht überlastet wird?
3. der Zustand des Fahrzeugs (codiert über schaltbare Widerstände im Fahrzeug, die die Spannung des PWM-Signals auf dem CP beeinflusst) detektiert? Nur im Status C (6V bzw. 1,3k Ohm) oder D (mit Belüftung. 3V bzw 270 Ohm darf das Schütz eingeschaltet werden, also der Stecker oder die Dose unter Spannung stehen.

snaptec, 2018-09-27:
- PP wird bei fest angeschlagenem Kabel durch Widerstand an der EVSE gesetzt
- CP wird immer an der EVSE angeschlossen

1. dies erkennt die EVSE.
OpenWB erkennt dies über die Auslesung des Ladeleistungszählers. Hierüber wird ebenso festgestellt mit wievielen Phasen gelesen wird.
2. Der Widerstand ist entsprechend des angeschlagenem Kabels zu wählen
Bei Nutzung mit Buchse ist PP mit der EVSE verbunden. Der Widerstand des Kabels (PP-PE) gibt der EVSE den maximalen Ladestrom vor.
3. Hierum kümmert sich (bewusst) die EVSE.
Der Raspberry macht nur "Vorgaben". So ist sichergestellt das wenn er mal Irr laufen sollte oder die SD Karte kaputt geht im schlimmsten Fall "nur" geladen wird oder eben nicht.

Die Anleitung der EVSE gibt ebenso Aufschluss:
http://evracing.cz/evse/evse-wallbox/evse-wb-din_20180611.pdf

bodems, 2018-09-27:
Ok, mir war nicht bewusst, dass der Pi "nur" Vorgaben macht und nicht den gesamten Ladevorgang überwacht und steuert. Das erklärt dann auch, warum ein einzelner DAC ausreicht.

closed 2018-09-27 by bodems