snaptec/openWB #22: Lizenz: GPL verbietet nicht kommerzielle Nutzung

mpollice, 2018-09-27:
In der Readme schreibt ihr "Die Software steht zur nicht kommerziellen Nutzung frei für jeden zur Verfügung.". Die GPLv3, die ihr als Lizenz verwendet, schließt kommerzielle Nutzung jedoch nicht aus, solange man sich an die Bedingungen hält (Bereitstellung jeglicher Veränderungen). Das mag für viele kommerzielle Anwendungen unerwünscht sein aber eben nicht unmöglich.

Falls ihr damit nur zum Ausdruck bringen wollt, dass ihr die Software zu kommerziellen Zwecken auf Anfrage auch unter einer permissiven Lizenz anbietet, dann wäre eine Klarstellung hilfreich. Wenn ihr tatsächlich die kommerzielle Nutzung eures Codes komplett ausschließen wollt, dann ist die GPLv3 die falsche Lizenz.

mpollice, 2018-09-27:
Lasst euch bitte von einem Anwalt beraten, wenn ihr tatsächlich die kommerzielle Nutzung ausschließen wollt. Ansonsten hängt es lediglich davon ab, ob jemand bereit ist, die Bedingungen der GPLv3 zu akzeptieren. Solange diese eingehalten wird, ist auch ohne Lizensierung eine kommerzielle Nutzung möglich.

snaptec, 2018-09-28:
Sehr unklar meinerseits ausgedrückt.
Ich hoffe nun ist es verständlich !?

closed 2018-09-28 by snaptec