snaptec/openWB #23: BHKW unterstützung

mr-gosh, 2018-09-27:
Wie sieht es mit der Unterstützung von Blockheitzkraftwerken aus? Auch hier ist eine intelligente Stromabnahme der Wallbox ein echter Gewinn an Funktionsumfang.

Ist das vorgesehen oder bereits möglich?

Wir planen bei uns dieses heir ein zu bauen:
https://www.blockheizkraftwerk.org/hersteller/ceramic-fuel-cells/bluegen

snaptec, 2018-09-27:
Da am EVU ausgelesen wird wieviel zur Verfügung steht, würde sich OpenWB den überschüssigen Strom auch in der derzeitigen Fassung "nehmen" und das Auto Laden.
Da das BHKW "nur" 1,5kW zur Verfügung stellt und die Minimalladung 6A bei einer Phase (je nach Auto) 1.200-1.400Watt benötigt, darf das Haus dann nicht viel verbrauchen.
Oder die PV ist unterstützend dabei.

Das Logging wird künftig noch überarbeitet. Hier wird ein BHKW dann separat (wie auch andere Energiequellen) einstellbar sein. Das dient aber mehr zur "Aufschlüsselung" des Nutzers.

Eine Funktion die seitens OpenWB das BHKW zum angehen triggert um zu unterstützten könnte ich mir allerdings vorstellen. Ist das eventuell so gemeint?

mr-gosh, 2018-10-22:
Es geht vor allem darum, dass ein BHKW ja wärmegetrieben ist und wenn Wärme gebraucht wird Strom als Nebenprodukt abfällt.
Wenn dieser Strom nicht gebraucht wird, wird er zu einem unattraktiven Kurs in Stromnetz eingespeist - daher wäre es super wenn sich so eine Wallbox die Diskrepanz zwischen Hergestellter und verbrauchter Energie schnappt und ins Auto einspeist.

Deluxe wäre dann die andere Richtung, die das Auto als Batterie fürs Haus verwendet, wenn mehr Strom gebraucht wird als durch das BHKW gerade hergestellt wird :-)

snaptec, 2018-10-22:
Das würde es allerdings jetzt schon tun im „nur pv“ modus.
Wenn Überschuss da ist - erstmal egal ob von PV oder BHKW - wird das Auto damit geladen.
Es muss nur genug Überschuss sein, sodass die Ladung beginnt.

Vehicle 2 Grid ist über AC leider nicht möglich.

snaptec, 2019-05-31:
Am Netzübergabepunkt ist entweder schon ein SmartMeter oder ein OpenWB EVU Kit nutzen.
Kam mit sonomotors etwas raus?

mr-gosh, 2019-05-31:
Die haben sich für das Feedback bedankt - mehr nicht. Da könnte vielleicht etwas mehr passieren, wenn die einer der die openWB Hardware verkauft / herstellt sich da mal meldet...

mr-gosh, 2019-05-31:
Bei unserem Netübergabepunkt ist ein Zweirichtungszähler lediglich mit "smart meter Vorbereitung".

snaptec, 2019-05-31:
Auf meine Anfragen gab es keine Reaktion. 
Wie immer bin ich den herstellern gegenüber aber offen wenn sie sich dann umentscheiden.

mr-gosh, 2019-06-12:
ich hatte bei dem ADAC Redakteur der den Test zu den Wallboxen geschrieben hat auch nachgefragt warum openWB nicht dabei ist - seine Antwort war auch etwas unbefriedigend.

lewie, 2019-06-18:
Hallo @mr-gosh,

naja, der ADAC ist ein erzkonservativer Verein. Den Herren ist das Open Source/Hardware Modell mindestens suspekt wenn nicht eher ein echter Dorn im Auge. OpenWB untergräbt wohl so einiges, wofür dieser Verein steht.

Ich kenne ja Dein Anschreiben nicht. Das es Fachleute sind, die eine openWB bauen, abnehmen und installieren. Oder selbst gebaute openWB abnehmen und genau auf die wichtigen Vorschriften hin überprüfen, scheint der ADAC wohl mit dieser Antwort in Abrede stellen zu wollen.

Aber ich möchte Dir widersprechen, die Antwort ist doch nicht unbefriedigend in Bezug darauf, dass einige wichtige Antworten darin enthalten sind. Der Antwortschreiber hat sich viel Mühe gegeben und seine Bedenken in Form von Hinweisen auf sehr hilfreiche einschlägige Richtlinienblätter ausgedrückt. All die Punkte kann man ja zB in die Empfehlungen für die Leute aufnehmen, die ihre openWB selbst montieren möchten.  

Und ich könnte mir vorstellen, zB über die Dekra Kontakte knüpfen, um openWB allgemein mal zu überprüfen. Das würde den Professionalisierungsgrad von openWB sicher erhöhen.

mr-gosh, 2019-06-19:
Vollkommen richtig - unbefriedigend war nur, dass die eigentliche Frage war, warum openWB nicht in den Test mit einbezogen wurde.
Ich hatte dann aber auch noch eine Nachfassmail geschickt um das zu konkretisieren...
Wenn es Dich interessiert kann ich Dir den Mail-Verkehr zukommen lassen. Aber das eigentlich interessante habe ich hier gepostet.

fritzchen-66, 2019-12-21:
Zum BHKW:
Ich würde versuchen, es stromgetrieben zu betreiben. Wärme läßt sich viel einfacher als Strom speichern.
Dann müßte die OpenWB Strom vom BHKW anfordern.

snaptec, 2019-12-21:
Hierfür kannst du dir die Smart Home / Webhooks zu nutze machen. 
Diese triggern bei überschreiten / unterschreiten inkl. mindesteinschaltdauer. 
Das muss dann nur anderweitig ausgewertet und das BHKW angesprochen werden. Bezweifle das die BHKWs sich per HTTP direkt steuern lassen!?

mr-gosh, 2019-12-25:
Bei einer Brennstoffzelle ist es ja andersherum.
Die produziert kontinuierlich Strom. (Bei mir 1,5kw/h)

Jetzt müsste ein script oder die wallbox selber einen smart meter bzw. eine Ausleseeinheit nutzen um zu entscheiden ob mehr Strom produziert wird als gebraucht wird, dann sollte diese überschüssige Menge ins Auto geleitet werden.

snaptec, 2019-12-25:
Wenn ein EVU Modul genutzt wird genau das ausgewertet. Der PV Modus erledigt das was du möchtest.
Ob das PV Strom ist oder nicht ist der openWB egal.
Sobald ein EVU Modul genutzt wird ist der PV Wert nur "für die Optik"

mr-gosh, 2019-12-25:
@snaptec 
ok - das klingt sinnvoll. Mal sehen wann sich der 2-richtungszähler bei mir remote auslesen lässt - die kommunikationsmodule sind leider bei unserem energieversorger noch inaktiv...

andig, 2020-05-17:
> Bei einer Brennstoffzelle ist es ja andersherum.
> Die produziert kontinuierlich Strom. (Bei mir 1,5kw/h)

Weil ich es so oft sehe: es sind kW*h, also kWh. Deine Leistung (ein Momentanwert) wird ja die ganze Stunde lang erzeugt.

mr-gosh, 2020-05-17:
> > Bei einer Brennstoffzelle ist es ja andersherum.
> > Die produziert kontinuierlich Strom. (Bei mir 1,5kw/h)
> 
> Weil ich es so oft sehe: es sind kW*h, also kWh. Deine Leistung (ein Momentanwert) wird ja die ganze Stunde lang erzeugt.

Da hast Du natürlich Recht - danke fürs klar stellen

MattvBeck, 2021-05-09:
Man könnte das BHKW als zweiten WR in openWB anlegen, dann stimmt die Bilanzierung. Hilfreich wäre in diesem Fall , dass die Umbenennung des zweiten „WR“ möglich wäre. Da eine BHKW nicht stark moduliert, genügt eigentlich nur der Zählerstand, der könnte dann in Leistung umgerechnet werden (oder umgekehrt mit dem Leistungswert)

mr-gosh, 2021-05-16:
Ich habe mittlerweile einen

closed 2025-07-20 by mr-gosh