snaptec/openWB #30: Einschränkungen bezüglich Fahrzeug

TheAlgorythm, 2018-10-01:
Ich habe in Foren gelesen, dass openWB nicht mit Teslas funktionieren soll.
Stimmt das?
Oder gibt es sonstige Einschränkungen bezüglich dem Fahrzeug?

snaptec, 2018-10-01:
Welches Forum verbreitet denn sowas?
Tesla lädt einwandfrei, bald ist sogar der SoC für Tesla in OpenWB sichtbar. 

Jedes EV das sich an den Ladestandard hält lädt ohne Probleme :)

TheAlgorythm, 2018-10-01:
Hätte mich auch gewundert.
Ein Kommentartor hier hat das gemeint: [Scheint aber für Tesla nicht zu funktionieren.](https://tff-forum.de/viewtopic.php?f=55&t=16680&p=518298)
Ok 👍, bis mein Model 3 kommt geht die SoC anzeige auch 😊.
Andernfalls kann ich versuchen das selber zu implementieren.
Vielleicht wäre es sinnvoll das in die Readme aufzunehmen ums klar zu machen:

> Jedes EV das sich an den Ladestandard hält lädt ohne Probleme

closed 2018-10-01 by TheAlgorythm