snaptec/openWB #89: Stromabrechnung/-logging: Ladungsfreigabe (go-e) - Stromkonten ?

BastianSalbert, 2019-02-08:
Lässt sich das go-e Modul so erweitern, dass die in der go-e App unter 'Zugangskontrolle' eingestellte Freigabemethode (Offen, RFID, Strompreise, Ladetimer) bei Betrieb mit der openwb Standalone erhalten bleibt?
Hintergrund ist der, dass an unserer go-e von verschiedenen Personen und zu verschiedenen Stromtarifen geladen wird (Person A Netz, Person A Solar, Person B Netz, Person B Solar, ...). Jede solche Kombination ist einem RFID Chip zugeordnet, der bisher zur Freigabe des Ladevorgangs gechipt werden musste, um die damit geladene Energie kumulativ zu speichern. Die geladenen kWh lesen wir dann über die go-e App aus und rechnen sie entsprechend ab.
Wenn wir die go-e allerdings mit der openwb betreiben startet der Ladevorgang sofort und wird keinen RFID Konto zugeordnet. Versuchen wir trotzdem zu chipen, treten Fehler an der go-e oder am zu ladenden Fahrzeug auf. Eine Stromabrechnung in der beschriebenen Art und Weise ist also zZ bei Betrieb der go-e mit openwb nicht möglich.
Um eine Abrechnung/Logging zu ermöglichen, müsste entweder die RFID Freigabe/Speicherung in der go-e erhalten bleiben, oder eine entsprechende Möglichkeit direkt in der openwb geschaffen werden. Eine Idee für zweitere Möglichkeit wäre, in openwb verschiedene Konten zu erstellen, wovon man vor jedem Ladevorgang das entsprechende auswählt (im Webinterface oder per neuem go-e/USB RFID Modul,...?), um die geladene Energie darauf zu speichern. Wenn man es ganz fancy machen will, kann man ja evtl direkt aus dem Konto heraus über PayPal zu einem hinterlegten Strompreis bezahlen.

snaptec, 2019-02-08:
Überlegungen gab es da natürlich schon.
Wie genau willst du das aber trennen?
Wenn Person A Netz sich autorisiert und lädt und du hast gerade 10kW PV Leistung ist das keine Ladung vom Netz.

Oder ist das nur für eine "interne" Abrechnung?

Was ich über die API auslesen kann ist "unlocked_by", hier wird die Nummer der RFID Karte mit übergeben.
Vermutung:
Hierüber könnte ich ein extra Log generieren das zum Ladevorgang noch die gescannte RFID mit ausliest.
Das müsste ich erstmal austesten.

Vor Ladevorgang auswählen ist blöd. Nach einer Weile Nutzung schaut man nicht mehr so oft ins Webinterface weil es automatisch geht.

BastianSalbert, 2019-02-08:
Okay, dann ist doch die einfachste und für mich völlig ausreichende Lösung, auf Freigabe der go-e zu warten und einfach direkt in der go-e den Strom zu zählen? Wenn jemand die go-e weiter ohne Freigabe betreiben will, stellt er halt in der go-e App "offen" unter "Zugangskontrolle" ein. Leider reichen meine Python Kenntnisse nicht aus als dass ich das selbst schnell implentieren könnte :-(
Eine Stromabrechnung in der openwb (zB wie von dir vorgeschlagen über Auslesung des RFID Namens) wäre dann nur ein Add On und für mich nicht notwendig.
PS: die Trennung/Zählung Netz/PV Energie ist nebensächlich und ein anderes Problem. Wir haben eine dreiphasige CEE Dose die vor der EVU Strommessung abzweigt, damit wir nicht den Hausakku ins Auto entladen. Umgekehrt haben wir eine einphasig angeschlossene CEE Dose an der Phase des Wechselrichters nach der EVU, die somit auch von der der Eigenverbrauchsregelung erfasst wird. Im Winter, bei schlechtem Wetter und für schnelle Ladungen nehmen wir die Netz Dose (Energie ~ nur Netz), bei PV Überschuss die solar Dose (Energie ~ nur PV). Entsprechend chipen wir dann, um den Strom den richtigen Abrechnungskonten zuzuordnen.

snaptec, 2019-07-27:
rfid erkennung ist in der beta

closed 2019-07-27 by snaptec