snaptec/openWB #122: Modules for Outlander PHEV, SMA Tripower 5.0 and SMA Sunny Island 6.0

roesl, 2019-04-08:
HI,

does anybody have modules for my devices or are existing modules compatible with my devices ?

regards
Georg

david-m-m, 2019-04-18:
The modbus protocol for all SMA tripower inverters is identical. You can safely use the tri900 module currently provided.

david-m-m, 2019-04-18:
Without having researched it: I'd think that the modbus protocol for Sunny Boy Storage and Sunny Island should also be identical. (At least the SBS module is using the standardized set of registers...)

snaptec, 2019-05-31:
Tripower can be used with die SMA WR Modul.
Sunny Island is integrated as modul in beta 1.49. They use slightly different registers.
Is there a api for the Outlander?
regards

snaptec, 2019-07-27:
As of no reaction

closed 2019-07-27 by snaptec