snaptec/openWB #131: Mindestüberschuss wird falsch berechnet

tammof, 2019-04-19:
Version 1.42
Mir scheint der Wert Mindestüberschuss aus den PV Ladeeinstellungen wird falsch berücksichtigt. Bei einem dreiphasigen Anschluss der go-e erscheint bei einem Eintrag von 
Mindestüberschuss: 4000 W
Folgender Eintrag im Log:
```
uberschuss 5705 wattbezug -5705 ladestatus 0 llsoll 0 pvwatt -6017 mindestuberschussphasen **12000** wattkombiniert 5705 abschaltungw 1560 schaltschwelle 690
Fri 19 Apr 13:20:45 CEST 2019
```
Die Ladung startet nicht, obwohl 5705 W Überschuss vorhanden sind 

Bei einem Eintrag von 
Mindestüberschuss: 1333 W
erscheint dann korrekterweise:

```
uberschuss 5729 wattbezug -5729 ladestatus 0 llsoll 0 pvwatt -6013 mindestuberschussphasen **3999** wattkombiniert 5729 abschaltungw 1560 schaltschwelle 690
Fri 19 Apr 13:20:55 CEST 2019
```
und die Ladung beginnt.

snaptec, 2019-04-19:
Absolut korrekt so. 
Die Werte werden immer für einphasiges Laden angegeben. 
OpenWB erkennt mit wieviele Phasen geladen wird und rechnet dann um. 
Ich ergänze den Hinweis in der Einstellungsseite.

closed 2019-04-19 by snaptec

tammof, 2019-04-19:
Gut zu wissen, danke!