snaptec/openWB #145: Number of phases is "nurpv.sh" and "minandpv.sh"

michaelstaemmler, 2019-05-20:
When I switch between 1 phase and 3 phase charging, the corresponding variable "anzahlphasen" switches like expected between "1" and "3".

Why do you use the number "4" as a distinction between both cases? E.g. for "if (( uberschuss < -1380 )); then" you write:
```
if (( anzahlphasen < 4 )); then
	llneu=$((llalt - 6 ))
else
	llneu=$((llalt - 2 ))
fi
```

I would have expected the number "3", because otherwise you always increase/decrease the charging current with the higher value (independent of the number of phases):
```
if (( anzahlphasen < 3 )); then
	llneu=$((llalt - 6 ))
else
	llneu=$((llalt - 2 ))
fi
```

What is not handled here is the case of 2 phase charging. So it would be better to just check if "anzahlphasen" is between 1 and 3, and then add/subtract "6/anzahlphasen", as this handles all three cases.

snaptec, 2019-05-31:
But there are even more cases (when you use more charging points)
As discussed on another channel, this is correct as is, but you are right. We could split that in exacter values for each amount of phases

snaptec, 2020-02-18:
fixed

closed 2020-02-18 by snaptec