snaptec/openWB #195: Anzeige "Aktuelle / Letzte Ladung" auf Startseite falsche Einheit

tobintax, 2019-06-13:
M.E. ist die Einheit von "Aktuelle / letzte Ladung" auf der Startseite nicht richtig. 

Geladen wurden z.B. 5 kwh. Angezeigt werden 5.000 kwh. 

Ist: 5.000 kwh
Soll: 5 kwh

snaptec, 2019-06-13:
Weil . statt , ?

tobintax, 2019-06-13:
richtig. 5.000 kwh entspräche 5 Megawattstunden (MWh). Vorschlag: Komma + lediglich 2 Stellen nach dem Komma.

snaptec, 2019-06-13:
2 Stellen ist kein Problem. Das , allerdings schon. Das Ladelog basiert auf ,-basierter CSV. Das müsste dann komplett umgebaut werden. Allgemein wird bei jedem Zählerstand etc.. Der Punkt als Trennzeichen genutzt. (auf der Status Seite auch gut ersichtlich)

tobintax, 2019-06-13:
ok, ich denke dass es mit 2 Stellen trotzdem aussagekräftiger wäre, auch ohne ",".

tobintax, 2019-06-13:
Analog auf status.php bei "Zählerstand in kWh"

LocutusB, 2019-06-16:
Ich denke es handelt sich ausschließlich um ein Problem zwischen Englischer und Deutscher Schreibweise. 5.000 soll 5,000 sein. Da aber auf der ganzen Oberfläche mit DezimalPUNKT gearbeitet wird, ist es zumindest konsistent ...

closed 2020-03-26 by snaptec