snaptec/openWB #208: Fronius Smartmeter und 2 Fronius Wechselrichter

Blacklimo1, 2019-06-19:
Hallo,
erst einmal bin ich absolut begeistert von openWB, toll das es solche Menschen gibt.
Ich habe eine Fronius Energy Meter für den EVU Übergabepunkt. Dieser ist an einem meiner beiden Fronius Wechselrichter dran.  Jetzt wird mir die aktuelle Leistung am EVU angezeigt und leider nur die PV Leistung von dem ersten Fronius WR. Es fehlt quasi die PV Leistung von 2ten Fronius WR.
Ist es möglich beide IP Adressen einzutragen, so das die Ladeleistung auch wirklich mit der gesamten PV Leistung berechnet wird.

Alternativ wäre auch ein Feld möglich wo ich eín Register von einem Modbus TCP Server eintrage, wo die aktuelle PV Leistung drin steht und openWB dann die Summe bildet. Wäre toll wenn man da was machen kann, da ich das sonst nicht wirklich effektiv nutzen kann. Vielen Dank

snaptec, 2019-06-19:
Da hat dein Solateur leider nicht ganze Arbeit geleistet.
Die Fronius WR werden normalerweise per DaisyChain verbunden.
Am WR 1 sind dann alle Daten der anderen WR abfragbar. OpenWB macht das so auch schon.

Wenn du allerdings die EVU Werte hast, ist der PV Wert nur für die Optik und das Logging. Die Regelung findet rein nach dem EVU Wert statt.

Blacklimo1, 2019-06-20:
Hallo Snaptec, danke für die schnelle Antwort. Der Solarteur war ich. 😉 Der 2 Wr steht 50m entfernt in einem anderen Gebäude. Beide sind am Heimnetzwerk derzeit liest meine SPS die Modbus Register der einzelnen WRs aus.  Kann Mann im Open Wb ModbusTCP Register zur Verfügung stellen wo ich dann die gesamte PV Leistung rein schreibe?

snaptec, 2019-06-20:
Wenn beide im selben Netzwerk sind kann man dem Haupt WR sagen "es gibt einen Slave per ModbusTCP erreichbar". Dann gibt der HauptWR alle Daten her.
Ein "blankes" ModbusTCP Modul gibt es derzeit nicht. Da es extrem viele Varianten zur Auslesung bzw. Speicherung der Werte gibt wäre das sehr aufwendig zu implementieren.
Wenn du die Werte auf einem HTTP Server ablegen kannst, kannst du das HTTP Modul nutzen.
Im Zweifel kannst du auch lokal ein kleines Skript schreiben das die Werte speichert. Z.B. Direkt in der Ramdisk der openWB. Im Anschluss per HTTP Modul lokal die Werte abholen.

Blacklimo1, 2019-06-20:
Hallo OK Danke, also die Modbus Slave Einstellung gibt es leider nicht beim Fronius. 
Ich werde mal schauen ob ich das überr den HTTP Server machen kann.

Blacklimo1, 2019-06-21:
Hallo Snaptec, 
über http hat es geklappt. Kannst du vielleicht  in der Dropdown Auswahl für den Ladepunkt(Ladeleistungszähler) auch die Möglichkeit einfügen die Daten über URL zu holen(also http), dann ist man noch unabhängiger und ich kann das auch gleich nutzen da ich die Daten schon im IoBroker habe. Das wäre super!!

snaptec, 2019-07-05:
Ladepunkt HTTP Modul ist drin.

closed 2019-07-05 by snaptec