snaptec/openWB #242: Nachtladen: Ignorieren des SOC bei veraltetem Zeitstempel

e-lmo, 2019-08-27:
Ich nutze EvNotify für das SOC-gesteuerte Nachtladen.
Das Smartphone für den Bluetooth-Empfang befindet sich in der Wallbox und nicht im Fahrzeug.
Daher wird der SOC nur während der Ladung und nicht beim Fahren aktualisiert.

Zum Zeitpunkt der nächsten Nachtladung ist der SOC aus Sicht von OpenWb größer oder gleich dem Ziel-SOC, wodurch die Nachtladung gar nicht erst gestartet wird.

Um dennoch die Nachtladung  automatisch starten zu können, wäre es nötig einen SOC mit Zeitstempel älter als z.B. 12 Stunden zu ignorieren

Alternativ könnte eine Aktualisierung des SOC auch beim Anstecken des Fahrzeugs getriggert werden.
Dazu wäre z.B. der Status "Fahrzeug angesteckt" auf LED3 oder ein Webhook ideal.

snaptec, 2019-08-28:
Der Zeitstempel ist je nach SoC Modul nicht unbedingt aussagekräftig.

Habe mal einen Webhook in 1.619 dafür hinzugefügt. Einstellbar in Einstellungen -> SmartHome.
Erfordert natürlich einen Ladepunkt der die Info "angesteckt" übermittelt.

Bitte um Feedback.

e-lmo, 2019-08-29:
Erste Rückmeldung:
Ich habe einen Sonoff POWR2 als Empfänger des Webhook eingetragen.
Neben dem Namen des Ladepunkts erscheint ein Steckersymbol auf der Hauptseite, aber der Webhook wird nicht ausgeführt.
Bei manueller Aktivierung des Webhooks (kopieren des Eintrags aus OpenWB in die Browser-Adresszeile) wird der SOC wie von mir gewünscht aktualisiert.

Mit dem Befehl PulseTime auf der Console des Sonoff kann man die Einschaltdauer begrenzen, sodass von OpenWB nur der Einschaltbefehl benötigt wird.

Mit freundlichem Gruß
Henri

snaptec, 2019-08-29:
Kleiner Bug drinnen gewesen, mit der 1.619b nochmal probieren bitte.

e-lmo, 2019-08-30:
Der Webhook funktioniert jetzt auch!
Echt klasse, wie Du auch auf solche speziellen Funktionen eingehst.
Ich werde nächste Nacht das komplette Zusammenspiel austesten.

snaptec, 2019-08-30:
Das kann man ja vielfältig einsetzen.
Wenn es z.B. eine Wärmepumpe oder andere Überschussverbraucher gibt kann man die per Webhook direkt deaktivieren

closed 2019-08-30 by snaptec