snaptec/openWB #290: USR-TCP232-304 als RTU/TCP Konverter?

andig, 2019-12-23:
Laut Doku sollte ein `USR-TCP232-410` funktionieren. Ich habe jetzt den "kleineren" `USR-TCP232-304`. Dieser bietet nur RS485, kein RS232 (trotz des Namens). Es gelingt mir aber partout nicht über den Adapter mit 2 angeschlossenen Zählern zu kommunizieren.

Parameter:

```
Baud Rate: 9600
Data Size: 8 Bit
Parity: None
Stop Bits: 1
Local Port Number: 26
Work Mode: TCP Server
```

Die Parameter funktionieren- wenn ich identisches Skript über RTU/USB Adapter laufen lasse kommen Daten, über TCP gibts nur i/o timeout; IP ist korrekt. Fehler des Adapters ist unwahrscheinlich da bereits das zweite Gerät.

Welche Tricks gibts da noch- Terminierung? Was könnte ich noch ausprobieren?

Zur Rerefenz das furchtbar einfache Skript in `go`:
```
package main

import (
	"log"
	"time"

	"github.com/grid-x/modbus"
)

type Logger struct{}

func (l *Logger) Printf(format string, v ...interface{}) {
	log.Printf(format, v...)
}

func main() {
	// handler := modbus.NewTCPClientHandler("rs485.fritz.box:23")
	handler := modbus.NewRTUClientHandler("/dev/ttyUSB0")
	handler.BaudRate = 9600
	handler.DataBits = 8
	handler.Parity = "N"
	handler.StopBits = 1

	client := modbus.NewClient(handler)
	handler.Logger = &Logger{}
	handler.SlaveID = 1

	for {
		if b, err := client.ReadInputRegisters(0, 2); err != nil {
			log.Println(err)
		} else {
			log.Printf("%02x", b)
		}

		handler.Close()
		time.Sleep(time.Second)
	}
}
```

snaptec, 2019-12-23:
welches Modul soll denn genutzt werden?

andig, 2019-12-23:
> welches Modul soll denn genutzt werden?

Soweit bin ich noch gar nicht- insofern Issue off-topic; ich versuche einfach nur den Adapter zum Leben zu erwecken.

> rs485.fritz.box:23 ?

Korrigiert- bei mir ists die 23. Telnet auf Port- IP als auch Name- funktioniert.

snaptec, 2019-12-23:
Achso, openWB macht eine Socket Verbindung auf wenn /dev/virtualcomX konfiguriert ist, dann lassen die sich wie herkömmliche lokale Adapter nutzen.
Beispiel:
`if [[ $sdm630modbusllsource = *virtual* ]]
then
	if ps ax |grep -v grep |grep "socat pty,link=$sdm630modbusllsource,raw tcp:$sdm630modbuslllanip:26" > /dev/null
	then
		echo "test" > /dev/null
	else
		sudo socat pty,link=$sdm630modbusllsource,raw tcp:$sdm630modbuslllanip:26 &
	fi
else
	echo "echo" > /dev/null
fi`
wenn $sdm630modbusllsource z.B. /dev/virtualcom0, dann kannst ihn dann lokal darunter ansprechen. Port 26 ist fix im Code, also bitte im USR umstellen.

andig, 2019-12-23:
Entschuldige- ich bin noch gar nicht bei openwb sondern noch "nackt". Heisst das, dass Du mit dem Adapter "Modbus RTU" sprechen musst anstatt "Modbus TCP" wenn Du ihn genauso behandelst wie ein lokales Device?

snaptec, 2019-12-23:
Richtig. die werden durch socat zum lokalen device. 
manche kann man umkonfigurieren.  die die beim evu kit / satelliten genutzt werden (anderer hersteller), werden direkt mit modbusTCP angesprochen.

closed 2019-12-23 by andig