snaptec/openWB #303: enhancement: flexiblere Zeiteinstellungen

tpd-opitz, 2020-01-11:
bisher sind alle Zeiteinstellungen select-Boxen mit der Granularität 1h. Das trifft aber nicht immer die (meine) Bedürfnisse. 

Ich wünsche mir die Zeiten mit höherer Auflösung einstellen zu können.

- Maximal-Lösung:
  <sub>Ohne auf die Konsequenzen in der Logik geachtet zu haben...</sub>
         `<input type="time" ... `

- Minimallösung:
  Aufnahme von Viertelstunden in die Auswahlbox.

snaptec, 2020-01-11:
Flexiblere Zeiteinstellungen bzgl. welchen Einstellungen?
Oft liegt es an der Logik dahinter die nur auf volle Stunden ausgelegt ist

tpd-opitz, 2020-01-11:
Ins besonderen das *Morgens Laden* wäre mir da wichtig. Ich weiß auf wenige Minuten genau wann ich morgens los fahre. Und in meinem Szenario benötige ich diesen Modus nur für die Standklimatisierung mit Netzstrom, also nur wenige Minuten. In meinem fall (Abfahrt X:30) lädt das Fahrzeug also 25Minuten zu lange...

snaptec, 2020-01-11:
Das Morgens Laden Feature ist in 15 Minuten Abständen konfigurierbar. (Beta/Nightly, noch nicht in der 1.6 Stable)

closed 2020-01-11 by snaptec