snaptec/openWB #304: feature request: Nacht-Laden: dynamische Ladeleistung

tpd-opitz, 2020-01-11:
Im Moduns *Nacht Laden* sollte die WB die Ladeleistung, die benötigt wird, um ausgehend vom aktuellen SOC in der eingestellten Nachtladezeit den Ziel-SOC zu erreichen, ausrechnen. Die Grenzen der Ladeleistung sollten von der Einstellung für *Sofort Laden* übernommen werden.

Wenn die errechneten Ladeleistung geringer als die eingestellte Untergrenze ist sollte das der Ladebeginn so lange verzögert werden, dass der SOC mit der minimalen Ladeleistung zur eingestellten  Zielzeit erreicht wird.

snaptec, 2020-01-11:
Das was das Zielladen macht?
Aber mit dauerhaft geringerer Ladeleistung wenn das Zeitfenster größer ist?

tpd-opitz, 2020-01-11:
Ja.

snaptec, 2020-01-11:
Es gibt aber durchaus viele Leute die das aufgrund geringeren Wirkungsgrades bei niedrigerer Ladeleistung nicht wollen.
Was wäre der Vorteil daran?

tpd-opitz, 2020-01-11:
der Vorteil wäre ehr für die Zukunft: so ein dynamische Modus läßt sich sicher besser an dynamische Strompreise anpassen, so dass die WB dann mehr Ladeleistung abgibt, wenn der Strom billig ist.

Ich denke, mein eigentliches Problem ist, dass Zielladen nur für den Ladepunkt 1 verfügbar ist und nicht mit den anderen Lademodi kombinierbar. Mein eigentlicher Request ist also Zielladen in das Nachtladen zu integrieren, zunächst  mit fester Ladaleistung.

snaptec, 2020-01-11:
Hier spielt dann wieder das Lastmanagement mit rein. Bei einem Ladepunkt ist das noch recht einfach. Bei mehreren und begrenztem Hausanschluss wird das Zielladen schnell komplex. Da ein großer Teil die openWB mit mehr als einem Ladepunkt betreibt ist das ein nicht unwichtiger Faktor.
Preisbasiertes Laden ist nochmal eine andere Baustelle und schon in Arbeit.

snaptec, 2020-02-18:
Preisbasiertes Laden ist implementiert

closed 2020-02-18 by snaptec