snaptec/openWB #306: soc_zoe: 'ZE services' zu 'My Renault' migriert  -> Kommunikation mit ZOE via "soc_zoe" nicht mehr möglich

ManuelBoss, 2020-01-11:
Renault hat 'ZE services' zu 'My Renault' migriert. Der Anmeldemechanismus hat geändert.
-> Kommunikation mit ZOE via "soc_zoe" ist nicht mehr möglich, Im GUI ist die Anzeige des Akkustandes nicht mehr möglich. Es wird nur "null % SoC" angezeigt.

Ein request wird wie folgt beantwortet:

request=$(curl -s -H "Content-Type: application/json" -X POST -d '{"username":"'$zoeusername'","password":"'$zoepasswort'"}' https://www.services.renault-ze.com/api/user/login)

echo $request
{"code":403900,"message":"com.worldline.renault.myzeonline.user.exception.RestrictedCountryException","timestamp":1578781259097,"redirectURLs":{"appstore":"https://myr-fr.renault.ch/ze-redirection.html?os=ios","playstore":"https://myr-fr.renault.ch/ze-redirection.html?os=android","website":"https://myr-fr.renault.ch/ze-redirection.html?os=web"},"locale":"de_DE"}

Weitere Details zum neuen API:
https://muscatoxblog.blogspot.com/2019/07/delving-into-renaults-new-api.html

snaptec, 2020-01-12:
Deutscher Account?

ManuelBoss, 2020-01-12:
Schweizer Account!

snaptec, 2020-01-12:
Ist hier schon Thema, derzeit ist schon Österreich und die Schweiz betroffen:
https://openwb.de/forum/viewtopic.php?p=5290

closed 2020-01-20 by snaptec