snaptec/openWB #323: LG changed API for LG ESS

sa0win, 2020-01-20:
Hi,
LG changed the structure of JSON API. I customized it for my system by hard coding the changes (as it was implemented in the current module implementation of bezug_lgessv1, wr_lgessv1 and speicher_lgessv1). My question/suggestion: 
I could implement a version of this modules which fits both APIs. For that I would insert a new pull down menu in the PV preferences section. Second Version: I also hard Code the new JSON API. All users would bei forced top update their systems.

What so You think?
BW
S.W.

snaptec, 2020-01-20:
set a flag f.e. lg_ess_api_vers and in case of 0 use old api, in case of 1 use new api in same module. 
There is already the same for Piko WR.

snaptec, 2020-01-20:
I would add that flag to modul config if that’s ok for you.

closed 2020-02-18 by snaptec