snaptec/openWB #386: Tippfehler in savesettings.php

benderl, 2020-02-18:
https://github.com/snaptec/openWB/blob/39c1176f89500d12a9681b2147296a16fcb728d5/web/tools/savesettings.php#L534

Die Zeile muss eigentlich so aussehen:
```
$result .= 'zielladenuhrzeitlp1=\''.$_POST[zielladenuhrzeitlp1]."\'\n";
```
Die vorletzten doppelten Anführungszeichen waren an der falschen Stelle.

snaptec, 2020-02-18:
Wie äußert sich das falsche Anführungszeichen?
Der Wert wird korrekt gespeichert, bei dir nicht?

benderl, 2020-02-18:
Ja, anscheinend wird es korrekt gespeichert, aber trotzdem ist die Syntax falsch. Nachdem ich meine WB gestern in Betrieb genommen habe schaue ich mir den Code näher an, um die unterschiedlichen Lademodi genau zu verstehen. In VS Code wurde direkt dieser Fehler angezeigt, daher dachte ich mir, ich melde es mal.
Mir sind noch ein paar nicht ganz saubere Codestellen aufgefallen, die auch Hinweise im Apache error.log erzeugen. Doch dazu kommt später noch eine Info. Ich muss mich erst einmal mit GIT vertraut machen, damit ich Dir ein Pull Request schicken kann.

closed 2020-03-24 by benderl