snaptec/openWB #421: enhancement: consistent logging graphs

tpd-opitz, 2020-03-26:
betrifft: Ver 1.7.302Nightly 

Die Darstellung der Grafen ist bei "Tag" anders als bei "Langzeitansicht".
[image]

[image]

Der Graf der "Langzeitansicht" ist für mich viel verständlicher. Bitte den Graf der Tagesansicht anpassen, so dass für die einzelnen Kurven die selben Daten wie in der Ansicht "Tag" genutzt werden

snaptec, 2020-03-26:
Das ist nicht so leicht möglich.
Die Langzeitansicht basiert auf Watt Werten.
Der Tagesgraph basiert auf Zählerständen.
Heißt bei Langzeit gibt es einen Wert Speicher, beim Tagesgraph hingegen gibt es Speicherimport & Speicherexport.

tpd-opitz, 2020-03-26:
Den Einwand verstehe ich nicht.  
Aus meiner (naiven) Sicht muss es doch möglich sein, beide Reports über die selben Daten laufen zu lassen, nur dass der Endpunkt im Tages-Report nicht mer relativ zur aktuellen Urzeit, sondern fix ist.

snaptec, 2020-03-29:
es werden aber unterschiedliche Daten zur Erzeugung herangezogen.

tpd-opitz, 2020-03-31:
Das sagtest Du bereits und ich bin ja auch darauf eingegangen. Die Frage ist: muss das so bleiben, wenn die "anderen Daten" keinen brauchbaren Graphen ergeben?

snaptec, 2020-03-31:
GGf. könnte man mit der Visualisierung herumspielen.
Am besten im Forum eine Umfrage starten ob Nachfrage vorhanden ist.

snaptec, 2020-04-04:
Problem mit DC seitigem Speicher gelöst

closed 2020-04-04 by snaptec