snaptec/openWB #422: bring back "interactive graphs"

tpd-opitz, 2020-03-26:
betrifft: Ver 1.7.302Nightly FireFox, Linux (Ubuntu)

In Version 1.6 waren die Darstellungen im Logging "interactive", man konnte also einen Breicht der Darstellung durch auswählen mit der Mouse zoomen und bekam (auch ohne zoom) an der Mausposition ein Fadenkreuz angezeigt, mit dem man die genaue Urzeit der Mouse-Position sowie die beiden dazugehörigen Skalenwerte angezeigt bekam. 

Bitte diese Funktionalität wieder aktivieren.

snaptec, 2020-03-26:
Die Graph Engine wurde geändert, da die alte doch zu imperformant war/ist.
Ob / wie gut das bei der neuen klappen gilt es zu testen, hat aktuell aber keine Priorität.

snaptec, 2020-04-04:
Zoom / Pan ist wieder implementiert in der Nightly

closed 2020-04-04 by snaptec