snaptec/openWB #469: 70% PV ist geil, aber...

theHolgi, 2020-04-14:
Hey, das habe ich auch gerade so gut wie implementiert, da poppt es im Hauptrepo auf... 😄 
Aber die Regelstrategie sollte angepaßt werden.

IST: Damit der Regler funktioniert, muss die Schwelle auf (MAX - 1400W) gesetzt werden, bei 10kW Peak also z.B. 5600W.
Bei > 7kW beginnt die Ladung, versucht danach aber eine Einspeisung von 5600W zu behalten.

SOLL:
Konfiguriert wird die maximale Einspeisung (d.h. 7000W oder knapp darunter).
Dieser Wert darf nicht überschritten werden. d.h.: 
- Starten der Ladung bzw. Erhöhung des Stroms **sobald die Schwelle überschritten wird**
- Verringern der Ladung bzw. Abschalten, wenn die erwartete Einspeisung dann unterhalb der Schwelle ist.
(Verringern: Einspeisung < Schwelle-230W*Phasen; Abschalten: Einspeisung < Schwelle-6A*230V*Phasen)

Ich schicke gerne einen PR dazu. In ein paar Tagen.

snaptec, 2020-04-14:
Ist das nicht das selbe wie jetzt nur das der mindestüberschuss auf 0 gesetzt ist? abschaltüberschuss muss ja bleiben, sonst macht er als hin und her

theHolgi, 2020-04-15:
Ja, das wäre wohl dasselbe - nur daß ich Mindestüberschuss ja für die "nur PV" - Ladung sinnvollerweise hoch setze (bloß nicht Strom kaufen) und mit dem 70% Laden (immer) außer Kraft setzen will.
Den Mindestüberschuss mit 0 zu überschreiben, wäre wohl die einfachste Implementierung.

snaptec, 2020-04-16:
Einzig die Berechnung des Abschaltüberschusses wirft bei mir noch fragen auf.
Sollte die nicht unterhalb der neuen Regelgrenze liegen?

theHolgi, 2020-04-16:
Die Abschaltung erfolgt mit dem Ausdruck:
`if wattbezugint > abschaltuberschuss`
Durch die Verschiebung des Nullpunktes arbeiten wir um 0 herum, in Richtung (virtuellem) Netzbezug (`wattbezugint` mit positiven Werten).
Wir können also dann abschalten (ohne die Schwelle sofort zu überschreiten), wenn der "Netzbezug" bei 6A*230V*Phasen liegt.

theHolgi, 2020-04-16:
Eliminierung eines der beiden Werte, "uberschuss" und "wattbezugint" wäre mir der Übersicht halber nicht ungelegen. Soll ich dazu auch mal einen PR machen?

snaptec, 2020-04-16:
argh, du verschiebst ja Wattbezugint mit, ich hatte "nur" Überschuss verschoben.
Ja bitte

theHolgi, 2020-04-17:
Nee, waren beide schon immer verschoben.
Eliminierung eines der beiden Werte machen wir aber nicht als Teil dieses PR, das würde ich dann neu im Master machen

PR #473 (2020-04-18): 70% Regler setzt eigene Regelschwellen
Fixes #469
closed 2020-04-18 by snaptec