snaptec/openWB #478: openwb-install.sh schlägt fehl

tho-wa, 2020-04-21:
Installation schlägt fehlt: (Raspbian Buster Lite)

pi@raspberrypi:~ $ curl -s https://raw.githubusercontent.com/snaptec/openWB/mast                              er/openwb-install.sh | sudo sh
update system
OK:1 http://archive.raspberrypi.org/debian buster InRelease
OK:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Paketlisten werden gelesen... Fertig
check for vim
Vormals nicht ausgewähltes Paket vim-runtime wird gewählt.
(Lese Datenbank ... 39847 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../vim-runtime_2%3a8.1.0875-5_all.deb ...
»Umleitung von /usr/share/vim/vim81/doc/help.txt zu /usr/share/vim/vim81/doc/hel                              p.txt.vim-tiny durch vim-runtime« wird hinzugefügt
»Umleitung von /usr/share/vim/vim81/doc/tags zu /usr/share/vim/vim81/doc/tags.vi                              m-tiny durch vim-runtime« wird hinzugefügt
Entpacken von vim-runtime (2:8.1.0875-5) ...
Vormals nicht ausgewähltes Paket vim wird gewählt.
Vorbereitung zum Entpacken von .../vim_2%3a8.1.0875-5_armhf.deb ...
Entpacken von vim (2:8.1.0875-5) ...
vim-runtime (2:8.1.0875-5) wird eingerichtet ...
vim (2:8.1.0875-5) wird eingerichtet ...
update-alternatives: /usr/bin/vim.basic wird verwendet, um /usr/bin/vim (vim) im                               automatischen Modus bereitzustellen
update-alternatives: /usr/bin/vim.basic wird verwendet, um /usr/bin/vimdiff (vim                              diff) im automatischen Modus bereitzustellen
update-alternatives: /usr/bin/vim.basic wird verwendet, um /usr/bin/rvim (rvim)                               im automatischen Modus bereitzustellen
update-alternatives: /usr/bin/vim.basic wird verwendet, um /usr/bin/rview (rview                              ) im automatischen Modus bereitzustellen
update-alternatives: /usr/bin/vim.basic wird verwendet, um /usr/bin/vi (vi) im a                              utomatischen Modus bereitzustellen
update-alternatives: /usr/bin/vim.basic wird verwendet, um /usr/bin/view (view)                               im automatischen Modus bereitzustellen
update-alternatives: /usr/bin/vim.basic wird verwendet, um /usr/bin/ex (ex) im a                              utomatischen Modus bereitzustellen
Trigger für man-db (2.8.5-2) werden verarbeitet ...
... installed
check for timezone

Current default time zone: 'Europe/London'
Local time is now:      Tue Apr 21 15:00:25 BST 2020.
Universal Time is now:  Tue Apr 21 14:00:25 UTC 2020.

...changed
check for bc
Vormals nicht ausgewähltes Paket bc wird gewählt.
(Lese Datenbank ... 41654 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../archives/bc_1.07.1-2_armhf.deb ...
Entpacken von bc (1.07.1-2) ...
bc (1.07.1-2) wird eingerichtet ...
Trigger für install-info (6.5.0.dfsg.1-4+b1) werden verarbeitet ...
Trigger für man-db (2.8.5-2) werden verarbeitet ...
...installed
check for apache
Vorkonfiguration der Pakete ...
Vormals nicht ausgewähltes Paket libapr1:armhf wird gewählt.
(Lese Datenbank ... 41672 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../00-libapr1_1.6.5-1_armhf.deb ...
Entpacken von libapr1:armhf (1.6.5-1) ...
Vormals nicht ausgewähltes Paket libaprutil1:armhf wird gewählt.
Vorbereitung zum Entpacken von .../01-libaprutil1_1.6.1-4_armhf.deb ...
Entpacken von libaprutil1:armhf (1.6.1-4) ...
Vormals nicht ausgewähltes Paket libaprutil1-dbd-sqlite3:armhf wird gewählt.
Vorbereitung zum Entpacken von .../02-libaprutil1-dbd-sqlite3_1.6.1-4_armhf.deb                               ...
Entpacken von libaprutil1-dbd-sqlite3:armhf (1.6.1-4) ...
Vormals nicht ausgewähltes Paket libaprutil1-ldap:armhf wird gewählt.
Vorbereitung zum Entpacken von .../03-libaprutil1-ldap_1.6.1-4_armhf.deb ...
Entpacken von libaprutil1-ldap:armhf (1.6.1-4) ...
Vormals nicht ausgewähltes Paket libbrotli1:armhf wird gewählt.
Vorbereitung zum Entpacken von .../04-libbrotli1_1.0.7-2_armhf.deb ...
Entpacken von libbrotli1:armhf (1.0.7-2) ...
Vormals nicht ausgewähltes Paket libjansson4:armhf wird gewählt.
Vorbereitung zum Entpacken von .../05-libjansson4_2.12-1_armhf.deb ...
Entpacken von libjansson4:armhf (2.12-1) ...
Vormals nicht ausgewähltes Paket liblua5.2-0:armhf wird gewählt.
Vorbereitung zum Entpacken von .../06-liblua5.2-0_5.2.4-1.1+b1_armhf.deb ...
Entpacken von liblua5.2-0:armhf (5.2.4-1.1+b1) ...
Vormals nicht ausgewähltes Paket apache2-bin wird gewählt.
Vorbereitung zum Entpacken von .../07-apache2-bin_2.4.38-3+deb10u3_armhf.deb ...
[... 503 lines cut ...]
Entpacken von php-gd (2:7.3+69) ...
libxpm4:armhf (1:3.5.12-1) wird eingerichtet ...
libjbig0:armhf (2.1-3.1+b2) wird eingerichtet ...
libwebp6:armhf (0.6.1-2) wird eingerichtet ...
fonts-dejavu-core (2.37-1) wird eingerichtet ...
libtiff5:armhf (4.1.0+git191117-2~deb10u1) wird eingerichtet ...
fontconfig-config (2.13.1-2) wird eingerichtet ...
libfontconfig1:armhf (2.13.1-2) wird eingerichtet ...
libgd3:armhf (2.2.5-5.2) wird eingerichtet ...
php7.3-gd (7.3.14-1~deb10u1) wird eingerichtet ...

Creating config file /etc/php/7.3/mods-available/gd.ini with new version
php-gd (2:7.3+69) wird eingerichtet ...
Trigger für libapache2-mod-php7.3 (7.3.14-1~deb10u1) werden verarbeitet ...
Trigger für man-db (2.8.5-2) werden verarbeitet ...
Trigger für libc-bin (2.28-10+rpi1) werden verarbeitet ...
E: Paket php7.0-xml kann nicht gefunden werden.
E: Mittels des Musters »php7.0-xml« konnte kein Paket gefunden werden.
E: Mittels regulärem Ausdruck »php7.0-xml« konnte kein Paket gefunden werden.
curl: (6) Could not resolve host: Upgrade
curl: (6) Could not resolve host: zu
error code: 1003curl: (6) Could not resolve host: Stable
curl: (3) URL using bad/illegal format or missing URL
pi@raspberrypi:~ $ kill: (12116): Kein passender Prozess gefunden
pi@raspberrypi:~ $

premultiply, 2020-04-21:
"7.0" durch "7.3" ersetzen.

tho-wa, 2020-04-21:
Danke. Jetzt habe ich noch ein paar Berechtigungsfehler im OpenWB Log: 

/var/www/html/openWB/runs/atreboot.sh: Zeile 15: /var/www/html/openWB/ramdisk/netzschutz: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 16: /var/www/html/openWB/ramdisk/hausverbrauch: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 17: /var/www/html/openWB/ramdisk/blockall: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 18: /var/www/html/openWB/ramdisk/llsoll: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 19: /var/www/html/openWB/ramdisk/ladungaktivlp1: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 20: /var/www/html/openWB/ramdisk/ladungaktivlp2: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 21: /var/www/html/openWB/ramdisk/ladungaktivlp3: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 22: /var/www/html/openWB/ramdisk/plugstat: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 23: /var/www/html/openWB/ramdisk/plugstats1: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 24: /var/www/html/openWB/ramdisk/chargestat: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 25: /var/www/html/openWB/ramdisk/chargestats1: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 26: /var/www/html/openWB/ramdisk/verbraucher1_watt: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 27: /var/www/html/openWB/ramdisk/verbraucher1_wh: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 28: /var/www/html/openWB/ramdisk/verbraucher2_watt: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 29: /var/www/html/openWB/ramdisk/verbraucher2_wh: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 30: /var/www/html/openWB/ramdisk/verbraucher2_whe: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 31: /var/www/html/openWB/ramdisk/verbraucher3_watt: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 32: /var/www/html/openWB/ramdisk/verbraucher3_wh: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 33: /var/www/html/openWB/ramdisk/verbraucher1_whe: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 34: /var/www/html/openWB/ramdisk/verbraucher1vorhanden: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 35: /var/www/html/openWB/ramdisk/verbraucher2vorhanden: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 36: /var/www/html/openWB/ramdisk/verbraucher3vorhanden: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 37: /var/www/html/openWB/ramdisk/evseausgelesen: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 38: /var/www/html/openWB/ramdisk/progevsedinlp1: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 39: /var/www/html/openWB/ramdisk/progevsedinlp2: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 40: /var/www/html/openWB/ramdisk/progevsedinlp12000: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 41: /var/www/html/openWB/ramdisk/progevsedinlp12007: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 42: /var/www/html/openWB/ramdisk/progevsedinlp22000: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 43: /var/www/html/openWB/ramdisk/progevsedinlp22007: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 44: /var/www/html/openWB/ramdisk/readtag: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 45: /var/www/html/openWB/ramdisk/rfidlp1: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 46: /var/www/html/openWB/ramdisk/rfidlp2: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 47: /var/www/html/openWB/ramdisk/rfidlasttag: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 48: /var/www/html/openWB/ramdisk/reloaddisplay: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 49: /var/www/html/openWB/ramdisk/ledstatus: Keine Berechtigung
/var/www/html/openWB/runs/atreboot.sh: Zeile 50: /var/www/html/openWB/ramdisk/execdisplay: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/wattbezug' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/ladestatus' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/lademodus' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/llaktuell' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/llaktuells1' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/pvwatt' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/soc' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/soc1' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/lla1' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/lla2' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/lla3' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/llkombiniert' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/llas11' kann nicht berÃ¼hrt werden: Keine Berechtigung
touch: '/var/www/html/openWB/ramdisk/llas12' kann nicht berÃ¼hrt werden: Keine Berechtigung
[... cut ...]
/var/www/html/openWB/runs/gsiabfrage.sh: Zeile 13: /var/www/html/openWB/ramdisk/gsiforecast.csv: Keine Berechtigung
/var/www/html/openWB/runs/gsiabfrage.sh: Zeile 13: /var/www/html/openWB/ramdisk/gsiforecast.csv: Keine Berechtigung
/var/www/html/openWB/runs/gsiabfrage.sh: Zeile 13: /var/www/html/openWB/ramdisk/gsiforecast.csv: Keine Berechtigung
chmod: Beim Setzen der Zugriffsrechte fÃ¼r '/var/www/html/openWB/ramdisk/gsiforecast.csv': Die Operation ist nicht erlaubt
kill: (2423): Kein passender Prozess gefunden
kill: (2424): Kein passender Prozess gefunden

tho-wa, 2020-04-21:
Besitzer Gruppe von ramdisk ist root/root. Gehört das so? 
drwxrwxrwx  2 root root  3460 Apr 21 16:45 ramdisk

benderl, 2020-04-21:
Nein, solle pi/pi sein.

tho-wa, 2020-04-21:
Umgestellt, aber nach Reboot gehört es wieder root/root.

snaptec, 2020-04-21:
Jessie nutzen, das wird aktuell auch genutzt. Migration zu neuerem OS folgt noch

closed 2020-05-18 by snaptec