snaptec/openWB #542: TWCManager - Fehlerhafte Ladestromvorgabe

Lennart-O, 2020-05-31:
Bei der Verwendung des TWCManagers an LP1 tritt ein Fehler bei der Stromregelung im PV Modus auf.

Bei Beginn der Ladung mit 6A werden diese auch korrekt an den TWCManager übergeben, dieser geht auf "Non scheduled power" 6A. Sobald openWB hochregelt wird allerdings der Strom nicht mehr als "non-scheduled power" auf z.B. 9 oder 10 A gesetzt, stattdessen wird auf "Track green energy" gestellt. Der TWC regelt nun auf bis zu 16A hoch, wodurch die Ladeleistung natürlich zu weit ansteigt. openWB regelt dann wieder auf 6A runter und das Spiel beginnt von vorn.

In welchem Teil des Codes wird diese Datenübergabe durchgeführt? Dann versuche ich das mal zu fixen.

snaptec, 2020-05-31:
https://github.com/snaptec/openWB/blob/master/runs/set-current.sh#L141

Sicher das das keine TWCManager Einstellung ist?
Bis dato gab es keine Bugreports zu der verwendeten Methode / Aufruf

Lennart-O, 2020-05-31:
Danke schonmal für den Link.

Ich habe einen Anhaltspunkt. Bei "Non-scheduled power" bietet das Webinterface vom TWCManager nur die Optionen 6, 8, 10, 13 oder 17A. Übermittelt openWB einen Wert der nicht in der dieser Liste ist, so springt TWC auf "Track green energy" und regelt auf 16A hoch. Übermittelt openWB einen Wert aus dieser Liste, so wird er korrekt übernommen.

Da muss man wohl was im TWC anpassen. Alternativ in openWB abfangen und den Strom entsprechend auf/abrunden, aber das ist natürlich Pfusch und die Regelungsqualität ist dahin.

Lennart-O, 2020-06-01:
Kann geschlossen/gelöscht werden. War ein Problem im TWCManager. Habe dort die standardmäßig verfügbaren Stromwerte ergänzt sowie die "Spike"-Regelung deaktiviert. Diese setzt der Ladestrom kurzzeitig hoch.

Fehlen nur noch stabile PV Daten von meinem SMA WR ;-)

closed 2020-06-02 by Lennart-O