snaptec/openWB #544: Tesla SoC Modul aktualisieren

snaptec, 2020-06-05:
Gewünscht ist der Umbau von der statischen Abfrage des SoC auf eine vorherige Prüfung des Zustandes (active/asleep).
Mit einem Flag muss es möglich sein das Fahrzeug dennoch gezielt zu wecken.
siehe:
https://openwb.de/forum/viewtopic.php?p=10202#p10193

benderl, 2020-07-10:
Wie im Forum bereits geschrieben, sehe ich mir das an und ändere auch gleich die Anmeldung auf das Token um. Das funktioniert soweit bei mir im Testbetrieb schon.

Es müsste jedoch noch folgender Ablauf integriert werden:

1. Erste Konfiguration des SoC-Moduls mit Benutzer, Passwort, Nummer
2. Beim ersten Aufruf des Moduls wird mit Benutzer und Passwort ein Token abgerufen
3. Das Passwort kann/muss jetzt gelöscht werden, sonst bietet die Variante auch nicht mehr Sicherheit
4. In der Konfiguration vom SoC-Modul sollte angezeigt werden, dass ein Token vorhanden ist. Eventuell könnte man es auch von dort direkt wieder löschen, wenn man das möchte.

Wie könnten die Punkte 3+4 aktuell umgesetzt werden? Wesentlich einfacher würde es natürlich, falls die Modulkonfiguration überarbeitet wird und man per Ajax mit der WB kommunizieren könnte. Ist das zufällig schon in Planung/Bearbeitung?

snaptec, 2020-07-10:
1:
Im Bestand vorhanden, bei Neukonfiguration muss es eingegeben werden
2:
Das könnte im SoC Modul selbst passiert.
3:
Das Passwort in der config wird nach erfolgreicher Token Generierung ersetzt mit z.B. tokeninuse, 2: kann prüfen ob Passwort != tokeninuse ist, wenn ja generiert es einen Token
4:
das wäre über Nutzung von tokeninuse einfach darstellbar. Wenn tokeninuse und das SoC "noch gültig" zurück gibt, ist kein Passwort einggebbar. Andernfalls natürlich schon. Das SoC Modul kümmert sich dann um Erstellung des Tokens und löscht das PW wieder. Es sei denn jemand benutzt das wirklich als Passwort.

Gerade bei Installation ist ggf. einiges "noch" holprig was das Netzwerk betrifft. Da macht es Sinn das Modul das ggf. 2-3 mal probieren zu lassen.
Wie lange ist das Token grundsätzlich gültig?

Angedacht ist es das wie die anderen Einstellungen auf MQTT umzusetzen. Gerade für die SoC Konfiguration soll dies aber nicht passieren. Die anderen SoC Module bieten diese Möglichkeit nämlich meist nicht. Passwörter sollen daher nicht auf den Broker wandern.

benderl, 2020-07-10:
> 1: Im Bestand vorhanden, bei Neukonfiguration muss es eingegeben werden
Ja, daran änder sich Nichts.
> 2: Das könnte im SoC Modul selbst passiert.
Das funktioniert bereits.
> 3: Das Passwort in der config wird nach erfolgreicher Token Generierung ersetzt mit z.B. tokeninuse, 2: kann prüfen ob Passwort != tokeninuse ist, wenn ja generiert es einen Token
Ok, das ist ein Weg.
> 4: das wäre über Nutzung von tokeninuse einfach darstellbar. Wenn tokeninuse und das SoC "noch gültig" zurück gibt, ist kein Passwort einggebbar. Andernfalls natürlich schon. Das SoC Modul kümmert [...]
Es gibt eine aktuellere Version von "teslajson.py", welche sich um das Token kümmert. Es werden zwei Tokens von Tesla zurückgeliefert, wenn man sich mit User/Pass anmeldet. Das erste ist für Abfragen gedacht, das zweite zum Anfordern eines neuen Tokens, wenn das erste bald abläuft. Das Script kümmert sich automatisch drum.
> Angedacht ist es das wie die anderen Einstellungen auf MQTT umzusetzen. Gerade für die SoC Konfiguration soll dies aber nicht passieren. Die anderen SoC Module bieten diese Möglichkeit nämlich meist nicht. Passwörter sollen daher nicht auf den Broker wandern.
Ja, bitte keine Zugangsdaten auf dem Broker. Ajax läuft ja nur zwischen Browser und Server und könnte zumindest etwas mehr Komfort bieten. Das ist dann für Version 2.xx.

benderl, 2020-07-13:
Der erste PR mit Umstellung auf Tokens ist drin: #572

snaptec, 2020-07-14:
Merged

closed 2020-07-14 by snaptec