snaptec/openWB #551: modules/bezug_fronius_sm/ anderen Systemaufruf verwenden

Moosbueffel, 2020-06-09:
Die Verwendung von 
`/solar_api/v1/GetPowerFlowRealtimeData.fcgi`
ist wesentlich performanter.

snaptec, 2020-06-10:
...Im Vergleich zu?

Moosbueffel, 2020-06-13:
Im Vergleich zu: `solar_api/v1/GetMeterRealtimeData.cgi?Scope=Device&DeviceID=0` Aber kommt halt darauf an, welche Daten man benötigt.

btw. Dieser Aufruf liefert bei mir einen Fehler. Die Devicenummer muss 1 sein!?

`Body	
Data	{}
Head	
RequestArguments	
DeviceClass	"Meter"
DeviceId	"0"
Scope	"Device"
Status	
Code	255
Reason	"Meter 0 does not exist"
UserMessage	""
Timestamp	"2020-06-13T19:22:42+0`

Mit DeviceID=1 bekomme ich gültige Daten. Oder halt Scope=System.

snaptec, 2020-06-17:
Ich vermute das Scope=System aber nur funktioniert wenn nur ein SmartMeter verbaut ist, richtig?

kaeferfreund, 2020-06-19:
```DeviceID=1``` ist wenn das Smartmeter als Bezugszähler verwendet wird. Bisher trage ich das nach jedem Update von Hand wieder ein. Wenn es ok ist würde ich das anpassen und einen PR machen @snaptec ?

kaeferfreund, 2020-06-19:
@Moosbueffel wenn ich das richtig sehe, dann wird in ```/solar_api/v1/GetPowerFlowRealtimeData.fcgi```der Bezug von Haus nicht angezeigt?

snaptec, 2020-06-26:
Ich bin aktuell mit Fronius in Kontakt um den besten Weg zu evaluieren.

michaelstaemmler, 2020-07-14:
Hallo zusammen,

ich habe das im (mittlerweile geschlossenen) Pull-Request von Manual nochmal kommentiert: https://github.com/snaptec/openWB/pull/570

@Kevin: hast Du schon was von Fronius gehört?

Viele Grüße,
Michael

kaeferfreund, 2020-09-22:
Hi @michaelstaemmler , 
danke für die ausführliche Stellungnahme! Ich suche jemanden der einen Fronius Hybrid/Gen24 hat und mir RS485 Nachrichten mitloggen kann zwischen Hybrid und Akku.

https://c2konzepte.de

kaeferfreund, 2020-09-26:
Der neue Gen24 WR ist montiert und die Variablennamen sind leider anders als bei der Symo Serie -.- Ich schreibe später mal ein kleines Modul dazu...

benderl, 2021-01-20:
Wie ist hier der aktuelle Stand? Gen24 sollte inzwischen funktionieren.

benderl, 2021-02-13:
Geschlossen, da keine Rückmeldung kommt.

closed 2021-02-13 by benderl

kaeferfreund, 2021-02-15:
Jau funktioniert alles gut