snaptec/openWB #554: Min+PV Laden startet nicht wenn in der Speicherbeachtung ein positiver Wert konfiguriert ist, im System aber kein Speicher angelegt wurde.

JanFellner, 2020-06-18:
Leider weicht die Beschreibung in den Einstellungen vom tatsächlichen Verhalten ab.

In den Einstellungen steht:
- Diese Funktionen sind nur aktiv, wenn ein Speicher konfiguriert wurde.

[image]

Dennoch startet das Laden Min+PV nur wenn beide Werte auf 0 gesellt sind.

Allgemein würde ich erwarten das Dinge, die keinen Einfluss haben, wahlweise nicht angezeigt werden oder inaktiv sind. Die Settings "erschlagen" einen leider wenn man sich initial mit openWB befasst...

snaptec, 2020-06-18:
Bin ich bei dir.
Erstmal, problem sollte gefixt sein - bitte in 1.8.005 prüfen.
Ausblendung steht noch auf der Agenda.
Ist stellenweise schon erfolgt, aber noch nicht 100% abgeschlossen

JanFellner, 2020-06-18:
I give it a try :)

JanFellner, 2020-06-18:
geht mit der 1.8.005 nightly kann zu.
Muss aber gleich noch eins aufreißen - Probleme und Verbesserungen für das Thema pushover .)

closed 2020-06-18 by JanFellner