snaptec/openWB #555: Support for Phoenix Contact EVCC

Maverick78de, 2020-06-18:
Please add support for Phoenix Contact EVCC Controller with modbus TCP and RS485
This Controller ist used e.g. in wallb-e Wallboxes.

Maverick78de, 2020-06-18:
If you need any documentation, I can provide it.

snaptec, 2020-06-18:
wallb-e doesn't want to be supported, I'm sorry.

Maverick78de, 2020-06-18:
I use the Phoenix Contact Controller which is open for buy, not only wallb-e use it! So it doesn't matter if wallb-e doesn't want to be supported.
e.g. https://esl-emobility.com/de/phoenix-contact-em-cp-pp-eth-2902802-zubehor-elektroinstallation-ladecontroller-ladesteuerung-wallbox.html

Maverick78de, 2020-06-18:
Modbus documentation for the most common controllers can be found here:
https://www.phoenixcontact.com/online/portal/de/?uri=pxc-oc-itemdetail:pid=1018701&library=dede&pcck=P-29-04-02-01&tab=1&selectedCategory=ALL
and here 
https://www.phoenixcontact.com/online/portal/de/?uri=pxc-oc-itemdetail:pid=2902802&library=dede&pcck=P-29-04-02-01&tab=1&selectedCategory=ALL

I can give you a summary if you can tell me which things are required for communication with openwb

snaptec, 2020-06-18:
I already contacted PhoenixContact if they would like to be integrated and will post the result here.

snaptec, 2020-06-26:
Except from auto respond mails we get no answer.
The problem:
- We implement EVCC
- A customer has a problem
- We have no support from Phoenix Contact
- Customer says "openWB has a crappy support"

We are absolutely fine with implementing that if we have a designated contact at PC to escalate problems in short time.

Maverick78de, 2020-06-26:
Ok .. Is it possible to implement a free configurable wallbox with http instead? So it could be possible to use any possible wallbox e.g. through ioBroker Rest API? You can mark this as unsupported method, so it could open openwb for more vendors.

snaptec, 2020-06-28:
Yes, that is an option to enable advanced users a simple possibility to use different hardware. 
HTTP (get) Requests or MQTT?

Maverick78de, 2020-06-28:
I personally would prefer HTTP GET which is easier to debug.

snaptec, 2020-06-28:
No problem. I will add that feature and also the possibility to ask „plugged“ status through http „Ladeleistungsmodul“ as this is essential for proper function

snaptec, 2020-06-29:
Please give it a try in 1.8.0.12 Nightly

Maverick78de, 2020-07-01:
It looks good for the first try, but could it be possible to define the whole url for the setting like in Ladeleistungmodul -> "HTTP" ?

[image]

andig, 2021-03-09:
@Maverick78de Phoenix it's implemented in evcc (although it has a known but at this time)

mi-hol, 2024-11-07:
> is implemented in evcc

@snaptec Does this mean this issue can be closed?