snaptec/openWB #556: Konfiguration von PushOver

JanFellner, 2020-06-18:
Habe Verständnisprobleme und damit vermutlich Verbesserungspotential für das Thema PushOver.

Habe die App gekauft, finde in der App ein dort so genannten "User key"
Diesen habe ich mangels wissens unter "User String:" eingetragen

Leider initial versehentlich unter dem App Token da die kontextuelle Beschreibung in dem UI Framework unter dem Edit Control liegt und damit (für mich) leicht verwirrend platziert. (Es würde Abstand zwischen Optionen und kontextueller Beschreibung fehlen um die Zugehörigkeit von Text zu Option klarer herauszustellen (persönliche Meinung))

[image]

Die Beschreibungstexte bringen hier imho keine verbsserung oder Klarheit da das was einzugeben ist ja schon ausreichend beschrieben wird.

Den App Token konnte ich nur durch auschecken der Sourcen in openwb.conf wieder finden.

- Der App Token sollte doch spezifisch sein für openwb. Warum sollte der Nutzer diesen editieren/ändern können? (Sollte man diese Option nicht ganz entfernen)
- Der User String sollte wie in der App und bei PushOver genannt User key heißen
- Ich vermisse einen Test Button. (Hab ich ehrlich gesagt auch in der Modul Konfig bei den ganzen 3. Systemen vermisst)

Habe die Funktion daher durch manuelles stoppen über die WebUI und reaktivieren des Ladens versucht zu triggern. Leider erfolglos. In den Logs finde ich hierzu nichts.
Wie kann ich das Problem am einfachsten diagnostizieren :)

snaptec, 2020-06-18:
> Der App Token sollte doch spezifisch sein für openwb. Warum sollte der Nutzer diesen editieren/ändern können? (Sollte man diese Option nicht ganz entfernen)
Der App Token ist der Token den der User in seiner Pushover APP erstellt.

JanFellner, 2020-06-18:
Schäm: Wer lesen kann ist klar im Vorteil 🤦 

Dennoch bekomme ich es nicht ans Laufen. In der WebUI von pushover kann ich mir als die eingerichtete APP eine Benachrichtigung schicken, diese wird auch zugestellt.

Auto ist leider voll, kann aktuell also nur anstecken abstecken als Push zustellen, was aber nicht gemeldet wird. In der WebUI verschwindet bzw. erscheint der Stecker beim abstecken anstecken.
Rebootet habe ich openWB ebenfalls bereits.

Gibts ein Log in openwb für die pushover Aktivitäten?

JanFellner, 2020-10-31:
Ich schließe das Ticket. Ein test button für die Einrichtung wäre Wünschenswert.

closed 2020-10-31 by JanFellner