snaptec/openWB #565: Verbraucher wir nicht im Graph dargestellt

Schildaero, 2020-07-06:
Hallo,
bei mir wird der Verbraucher im Graphen nicht dargestellt (Und in der Übersicht steht "Verbraucher1" und nicht wie erwartet der eingetragene Name, aber das wäre unwichtig).
[image]

Was ich bisher gesehen habe ist, dass über MQTT die Werte im topic "Verbraucher" richtig übertragen werden (siehe Bild).
[image]

 Aber nicht in den alllivevalues in mqtt:
08:26:44,-198,0,2876,0,0,0,,,0,,2678,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0

Wenn ich die live.js richtig interpretiert habe, würde der Verbraucher in dem csv hinter dem Hausverbrauch stehen, hier also hinter der 2678. 
Die all-live.graph im ramdisk sieht genauso aus. 
Die Datei verbraucher1-lice.graph ist angelegt aber leer.

Vielleicht als Idee wo das Problem liegen könnte: Da mqtt die Werte anzeigt, stimmt ja die Kommunikation, meine Vermutung ist, dass die Variable $verbraucher1_watt nicht befüllt wird, und daher nicht im alllive landet. Leider habe ich bisher die Stelle im Code nicht gefunden, wo das passiert (oder eben nicht), sonst hätte ich gerne schon eine Lösung präsentiert.

Vielleicht gibt es noch ein Problem durch mein Setup, dass irgendwo ein Bit falsch abbiegt, daher hier noch eine kurze Beschreibung:
Ich werden ein SDM220, das mittels USB-RS485 an einen separaten Raspberry angeschlossen ist, und dieses eine RTU-TCP Modbus bridge auf den port 8899 herstellt. In openWB habe ich Verbraucher 1 als SDM120 konfiguriert, da ich sowohl in den Handbüchern als auch im Code gesehen habe, dass die die gleichen Register für den Verbrauch verwenden.

Ansonsten auch noch einmal vielen Dank an openWB für die Idee an sich, und vor allem dass es open source ist. Und vielen Dank für den Support.

VG
Marco

snaptec, 2020-07-06:
Bitte mal ein Update auf Nightly 1.8.101 machen, sollte nun ersichtlich sein.

Schildaero, 2020-07-06:
Ja es funktioniert, besten Dank.

closed 2020-07-06 by Schildaero