snaptec/openWB #566: SmartHome Verbraucher

kaeferfreund, 2020-07-07:
Moin,
ich finds echt genial, dass es eine solche Funktion bereits gibt! Leider wird mein SmartMeter noch nicht unterstützt. 
Bei dem Versuch dieses als SmartHome Verbraucher zu implementieren stellt sich mir die Frage ob diese Funktion in der aktuellen nightly überhaupt aktiv ist?

Schildaero, 2020-07-07:
Moin,
Ich hatte gestern dazu einen Issue aufgemacht, das wurde bereits gestern behoben und ist in der nightly vorhanden. Welches Smart Meter hast du denn? Wenn du eine IP Adresse bei Verbraucher eingibst, dann wird (soweit ich das gesehen habe) Versucht auf den TCP Port 8899 zu verbinden (ist wohl ein gängier port für das im Shop angebotene RTU (Serialles Kabel) zu TCP Gateway. Einige Smart Meter hören für Modbus lieber auf den port 502. Also kann es schonmal sein, dass openWB dein SM gar nicht kontaktieren kann.
Und selbst wenn, ist es noch wichtig, wie die Verbrauchsdaten Addressiert sind. Heißt, selbst wenn die Ports stimmen, kann es sein, dass die Informationen nicht gefunden werden können.
Daher die Frage: welches SM hast du? :)

VG
Marco

Schildaero, 2020-07-07:
#565 War die Nummer, fürs Protokoll

kaeferfreund, 2020-07-07:
ABB-B23 ist das SmartMeter, ist mit einem USR-TCP232-304 angeschlossen. Über Python kann ich das soweit ansteuern und auslesen, das klappt.

[2CMC485003M0201_A_en_B23_B24_User_Manual.pdf-2.pdf](https://github.com/snaptec/openWB/files/4884201/2CMC485003M0201_A_en_B23_B24_User_Manual.pdf-2.pdf)

Schildaero, 2020-07-07:
`Über Python kann ich das soweit ansteuern und auslesen, das klappt.`
Das klingt doch schon einmal prinzipiell gut. Danke für das Manual. Mit Python meinst du da mit dem script von openWB, oder ein eigenes? 
Wenn du ein eigenes hast, nutzt du Modbus oder M-Bus Protkoll?
Für mich sehen die Register auf dem SDM630 Manual sehr verschieden zu denen vom ABB aus. Aus dem Bauch raus würde ich sagen, wenn du die register im script anpasst, der tcp port mit dem vom USR-TCP übereinstimmt, könnte es vielleicht funktionieren.

USR-TCP232-304 Manual:
https://www.usriot.com/download/M0/USR-TCP232-304-User-Manual_V1.0.3.01.pdf

kaeferfreund, 2020-07-07:
```from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.transaction import ModbusRtuFramer
import ctypes

client = ModbusClient('192.168.1.51', port=502, framer=ModbusRtuFramer)
client.connect()
response = client.read_holding_registers(address=0x5B00, count=66, unit=int(2))
print(response)
voltage = ((response.registers[0] << 16) | response.registers[1]) / 10
activePowerTotal = ((response.registers[20] << 16) | response.registers[21])
print("Voltage L1-N: " + str(voltage) + "V")
print("Active Power Total: "
              + str(ctypes.c_int32(activePowerTotal).value / 100)
              + "W (" + hex(activePowerTotal) + ")")
response = client.read_holding_registers(address=0x5000, count=4, unit=int(2))
activeImport = (
            (response.registers[0] << 32)
            | (response.registers[1] << 24)
            | (response.registers[2] << 16)
            | response.registers[3]) / 100
print("Imported: " + str(activeImport) + "kWh")
client.close()```

Config vom USR poste ich gleich auch noch fix. Das ist alles kein Problem.

Ich bin dabei gerade die passende Implementierung zu schreiben, einziges Hindernis ist nur, dass ich einfach nichts auf dem Status oder HomeBildschirm angezeigt bekomme auch wenn ich im Code für den Verbraucher ```SDM630remote.py``` einen festen Wert hinterlege, der dann in die Ramdisk geschrieben wird. 

Da es grundsätzlich funktioniert, weiß ich jetzt schonmal, wo ich nach meinem Fehler suchen muss.

kaeferfreund, 2020-07-07:
So! Eine Runde weiter: In der smarthome.php steht in Zeile 540 ``` Modbus ID.``` - aber ich sehe weder im Browser noch im Quellcode eine Möglichkeit die Modbus ID festzulegen? Übersehe ich etwas?

Schildaero, 2020-07-07:
`Allerdings wird die SDM630remote.py bei mir nicht automatisch ausgeführt. Jemand eine Idee woran das liegen könnte?`
Du meinst damit, dass das ganze Skript nicht aufgerufen wird? In der hook.sh werden die Skripte aufgerufen, so etwa bei Zeile 160. Die Logik für das SDM630 sieht identisch mit dem SDM120 aus, was ja bei mir funktioniert, daher würde ich erstmal diese Stelle ausschließen
Da werden auch die IDs übergeben. 
`python modules/verbraucher/sdm630remote.py 1 $verbraucher1_source $verbraucher1_id`
(Die Modbus ID nicht mit der "Verbraucher" ID, also ob es der erste oder der zweite Verbraucher ist, verwechseln.) Es muss ja am Ende eine Datei verbraucher1_wh in ramdisk herauskommen, die dann später wieder eingelesen wird, zum Versenden zur Grafik.
Hoffe, das hilft weiter.

kaeferfreund, 2020-07-07:
Jau danke! Das klappt soweit. Irgendwie wir bei mir die hooks() Funktion nicht aufgerufen. Soweit bin ich inwzischen. Wenn ich die per Befehl aufrufe klappt alles wie ich es gerne hätte. 
Muss gleich mal schauen unter welchen Bedingungen die hooks() Funktion aufgerufen wird

Schildaero, 2020-07-07:
Wenn ich es richtig sehe in regel.sh Zeile 172

kaeferfreund, 2020-07-07:
Jau! Danke für die schnelle Hilfe. Ich habe das ganze jetzt Eingrenzen können und es ist etwas strange:
Ich habe eben die Daten auf den aktuellen Stand des Repos zurückgesetzt und nur die angepasste Datei ```sdm630remote.py```ersetzt. Mit einem Verbraucher läuft das ganze einwandfrei. Bei zwei Verbrauchern scheint es zu Problemen bzgl. dem Polling zu kommen, scheinbar kommt der URS oder die Modbus Geräte nicht hinterher und der TimeOut schlägt zu. Ich poste gleich mal die Einstellungen von dem USR.

Hat jemand da eine Idee?

kaeferfreund, 2020-07-08:
Anscheinend war das Polling einfach zu schnell - Mit nem sleep .2 geht es dann. Schaue gleich nochmal in wie weit das andere Prozesse blockiert und dann kommt der PR

Schildaero, 2020-07-08:
Das klingt doch gut. Wenn ich das richtig sehe, ist in der hook.sh beim Verbraucher2 noch nicht vorhanden, dass von ramdisk die werte in variablen gelesen werden (um dann auch in der Grafik dargestellt zu werden). Siehst du das auch so? Falls ja, wenn du eh einen PR machst, magst du das Vielleicht gleich mit testen, ergänzen und reinnehmen?

kaeferfreund, 2020-07-08:
Jau, das hatte ich auch gesehen und behoben^^

kaeferfreund, 2020-07-08:
#569

closed 2020-07-08 by kaeferfreund