snaptec/openWB #571: Graph nach Neustart

kaeferfreund, 2020-07-12:
Moin!
ich betreibe die openWB auf einem Ubuntu Server (der als vServer bei mir läuft). Nach dem Neustart des Server habe ich immer das Problem, dass anscheinend ein Skript nicht gestartet wird. Hat noch jemand das Problem?

Wenn ich dann ein Update mache, läuft alles einwandfrei...

Vermute mal ein Shell Skript wird nach dem Neustart nicht gestartet?

snaptec, 2020-07-12:
runs/atreboot.sh wird beim neustart / nach update ausgeführt.

kaeferfreund, 2020-07-12:
Top, Danke! Im crontab stand nichts drin. Muss ich nochmal checken, warum das so gelaufen ist bei der Installation.

Falls jemand das gleiche Problem hat:

```crontab -e```
Da sollte dann ```@reboot /var/www/html/openWB/runs/atreboot.sh```drin stehen, falls nicht nachtragen