snaptec/openWB #574: Tesla SoC Modul nach Update funktionsuntüchtig

snaptec, 2020-07-14:
`/var/www/html/openWB/modules/soc_tesla/teslajson.py:123: UserWarning: Could not open file /var/www/html/openWB/modules/soc_tesla/tokens.lp1: [Errno 2] No such file or directory: '/var/www/html/openWB/modules/soc_tesla/tokens.lp1' (pressing on in hopes of alternate authenticaiton)
  warnings.warn("Could not open file %s: %s (pressing on in hopes of alternate authenticaiton)"%(self.tokens_file, str(e)))
Traceback (most recent call last):
  File "/var/www/html/openWB/modules/soc_tesla/teslajson.py", line 348, in <module>
    c = Connection(email=args.email, password=args.password, access_token=args.access_token, tokens_file=args.tokens_file, tokens_file_use=args.tokens_file_use, proxy_url=args.proxy_url, proxy_user=args.proxy_user, proxy_password=args.proxy_password, retries=args.retries, retry_delay=args.retry_delay, debug=args.debug)
  File "/var/www/html/openWB/modules/soc_tesla/teslajson.py", line 125, in __init__
    self.vehicles = [Vehicle(v, self) for v in sorted(self.get('vehicles')['response'], key=lambda d: d['id'])]
  File "/var/www/html/openWB/modules/soc_tesla/teslajson.py", line 131, in get
    return self.post(command, None)
  File "/var/www/html/openWB/modules/soc_tesla/teslajson.py", line 138, in post
    self._refresh_token()
  File "/var/www/html/openWB/modules/soc_tesla/teslajson.py", line 185, in _refresh_token
    tokens = self.__open("/oauth/token", data=self.oauth)
  File "/var/www/html/openWB/modules/soc_tesla/teslajson.py", line 239, in __open
    raise last_except
urllib2.HTTPError: HTTP Error 401: Unauthorized
`

Tesla Logfile (ramdisk/soc-tesla.log) wird nicht erstellt.

benderl, 2020-07-14:
Das Log wird erstellt, wenn debug gesetzt ist.
Die Meldung "No such file..." ist normal und nicht schlimm, da die beim ersten Mal erstellt wird.
"HTTP Error 401: Unauthorized" deutet darauf hin, dass es ein Problem mit den Zugangsdaten besteht. Ich hatte es einmal beim Testen, dass wegen zu vieler ungültiger Anmeldeversuche der Account gesperrt wurde.
Am Besten in dem Fall:
1. die Zugangsdaten in openWB entfernen
2. auf der Tesla Webseite einloggen
3. jetzt wird man darauf hingewiesen, dass der Account gesperrt wurde und man das Kennwort zurücksetzen soll
4. neues Kennwort gemäß Anleitung setzen
5. Zugangsdaten neu in openWB eintragen

benderl, 2020-07-14:
Mit dem PR #575 wird die Situation etwas entschärft, da nicht in jeder Regelschleife die Tesla API angesprochen wird.
Besser wäre natürlich, wenn das Modul den Benutzer irgendwie über ein falsches Passwort informieren könnte und die Zugriffe dann unterbunden werden bevor der Account gesperrt wird.

closed 2020-07-14 by snaptec