snaptec/openWB #584: E3DC  mit go-echarger und Tesla, Ladeleistung wird nicht mehr angezeigt

Sonnenstupsi, 2020-07-26:
[image]

Seit einigen Tagen wird die Ladeleistung nicht mehr angezeigt, obwohl der Tesla hier zB mit 6A lädt. Die Überschussladung (PV-Laden)  funktioniert damit auch nicht. Es wird immer mit dem am go-echarger eingestellten A-Wert geladen.
Auch das Umschalten von PV-Laden auf Stop wird komplett ignoriert.
Zunächst konnte ich den einwandfreien Betrieb durch ein System-Reboot wiederherstellen aber das funktioniert jetzt auch nicht mehr!
OpenWB-Version Stable: 1.8.000StableRC

snaptec, 2020-07-27:
Vermutlich hat sich die go-e IP geändert oder der go-e ist nicht mehr im WLAN

Sonnenstupsi, 2020-07-27:
Ja, da wäre ich nicht drauf gekommen, die IP Adresse vom Go echarger hatte sich geändert. Danke für den Tipp!Von Samsung-Tablet gesendet

https://github.com/snaptec/openWB/issues/584#issuecomment-664188316

closed 2020-07-31 by snaptec