snaptec/openWB #591: Installation von openwb auf raspbian schlägt fehl.

etechnik3r, 2020-08-08:
Wenn ich auf einem frischen rasbian openwb über
`curl -s https://raw.githubusercontent.com/snaptec/openWB/master/openwb-install.sh | sudo sh`
installieren möchte, dann bricht die Installation mit folgendem Fehler ab:

```
ln: failed to create symbolic link '/var/www/html/openWB/ramdisk/openWB.log': File exists
...changed
evdev installed...
curl: (6) Could not resolve host: Upgrade
curl: (6) Could not resolve host: zu
error code: 1003curl: (6) Could not resolve host: Stable
curl: (3) URL using bad/illegal format or missing URL
```

Wo könnte hier das Problem liegen?

Rutschii, 2020-08-15:
Hallo,
Habe gestern ein neues Raspi Image erstellt und openWB installiert, bei mir hat es funktioniert. 

Hab den offiziellen Raspi Imager mit dem empfohlenen Standard Image genommen. 

Evtl SSH in der Raspi config aktivieren so wie in der Anleitung beschrieben. 

Grüße

etechnik3r, 2020-08-20:
> Hallo,
> Habe gestern ein neues Raspi Image erstellt und openWB installiert, bei mir hat es funktioniert. 
> Hab den offiziellen Raspi Imager mit dem empfohlenen Standard Image genommen.
 
> Evtl SSH in der Raspi config aktivieren so wie in der Anleitung beschrieben.

Danke für deine Antwort. Habe ich genau so probiert, leider noch immer das gleiche Problem.

etechnik3r, 2020-09-02:
Scheinbar führt die Zeile
`curl -d "update="$releasetrain$uuid"vers"$owbv"" -H "Content-Type: application/x-www-form-urlencoded" -X POST https://openwb.de/tools/update.php` in atreboot.sh zu einer Fehlermeldung.
Mehrere " machen hier wohl Probleme?

snaptec, 2020-09-05:
Nein, das ist nicht ursächlich - siehe Forum

closed 2020-09-05 by snaptec