snaptec/openWB #599: Audi SOC Modul geht nicht mehr

EyeOfTheStorm, 2020-08-14:
Audi hat scheinbar die API vor ein paar Tagen geändert. Kann sich das mal jemand ansehen?

andig, 2020-08-16:
Siehe https://github.com/davidgiga1993/AudiAPI/issues/13

andig, 2020-08-26:
Es braucht mal ein Call Trace der App. Im Ticket steht wie das geht- ich habe leider kein Android...

premultiply, 2020-08-26:
Gibts die App nicht als APK welches man einfach entpacken kann und dann den Quelltext ansehen kann?
Das ist doch meist ohnehin so ein zusammengeschusterter Web-Javascript-Java-Baukasten-Kram

Normalerweise geht das bei Android-Apps

andig, 2020-08-26:
Genau, steht alles unter dem Link. Ich hab kein Android...

premultiply, 2020-08-26:
Ich auch nicht. Braucht man aber auch nicht. Dafür gibt es Internetdienste denen man das runtergeladene APK einwirft und dann den entpackten Salat bekommt.

stilicrafter, 2020-10-11:
Google stellt doch einen android emulator zur Verfügung.

andig, 2021-03-11:
Das Problem ist lange gelöst, Issue kann geshclossen werden.

andig, 2021-03-17:
Was ist an diesem Kommentar off-topic in Anbetracht von https://github.com/snaptec/openWB/commit/79a1563ee2913382d4dcf12315d224f5f54f822f?

closed 2021-03-17 by benderl