snaptec/openWB #602: Umrechnung EVU Power Faktor bei Einspeisung falsch

miradarya, 2020-09-03:
Auf der Status-Seite wird der EVU Power Faktor falsch angezeigt, wenn Einspeisung vorliegt. Korrekt müsste dann ein negativer Wert zwischen -1 und 0 angezeigt werden, es wird aber ein Wert 5,... angezeigt. Ich vermute, das liegt daran, dass resp.registers[1] fehlerhaft als 16Bit-unsigned-Wert interpretiert wird, der Zähler liefert aber einen 16Bit-signed-Wert:

https://github.com/snaptec/openWB/blob/82e3b35f73f2c77100c141b77806ca9f01c87462/modules/bezug_ethmpm3pm/readlovato.py#L123

Ich bin aber kein Python-Experte und kann es mangels ssh-Zugriff auf meine openWB auch nicht testen.

kevinwieland, 2020-09-03:
Wird geprüft.

benderl, 2022-07-27:
@LKuemmel kannst Du bitte die aktuelle Umsetzung prüfen?

miradarya, 2022-07-27:
Sorry, ich glaube, ich hatte vergessen den issue in der Zwischenzeit zu schließen. Ich meine, dass der Fehler mittlerweile behoben ist. Bin momentan im Urlaub, aber ich kann das auch gerne prüfen, wenn ich zurück bin.

LKuemmel, 2022-07-28:
Der Wert wird als INT_32 ausgelesen. Lt Doku ist der Wert 2 Worte lang, also 32 Bit.

miradarya, 2022-07-31:
Fehler tritt in der aktuellen Version nicht mehr auf.

closed 2022-07-31 by miradarya