snaptec/openWB #625: OpenWB lädt, ich kann aktuell faktisch nur das Auto abstecken

JanFellner, 2020-10-02:
Leider vermisse ich in der WebUI immer noch eine Möglichkeit zu diagnostizieren warum openWB macht was es macht.

Ohne jegliche Eingriffe (Confi/Firmware) hat OpenWB gestern einfach weiter geladen obwohl keinner der dafür nötigen Faktoren erfüllt war.

- PV hat nichts mehr geliefert.
- Speicher wurde entladen.
- Die Abschaltverzögerung war komplett überschritten.

Ich habe auf Stop gestellt -> Lädt weiter. Ich habe also das Auto abgesteckt.

Heute früh neuer Versuch. Selbes Verhalten. Obwohl keiner der Faktoren ein Laden erlauben dürfte wird angefangen zu Laden.

LadeController ist ein EVSE-Wifi Version 1.02 konfiguriert auf Remote Controlled.
Im Register 1000 steht 6 obwohl openWB auf STOP steht.

Wie kann ich sowas als Nutzer irgendwie diganostizieren?
Hatte jetzt alle möglichen Versionen drauf:
Stable: 1.8.000StableRC (aktuell drauf)
Beta: 1.8.400Beta 
Nightly: 1.8.406Nightly

JanFellner, 2020-10-02:
Vom Raspi (virtualisiert auf ESXI) komme ich zum EVSE.
Was passiert wenn openwb den LadeController temporär mal nicht erreichen kann (hängt ja im WLAN).
Wird das nochmals versucht?

-> insgesamt fehlt mir irgendwie ein technisches Log um bei Problemen zunächst mal selbst schauen zu können.
Gibt es sowas wenn ja wie kommt man dorthin? In der WebUI finde ich nichts um das Log abzugreifen.

[image]

snaptec, 2020-10-02:
das Log ist unter /var/www/html/openWB/ramdisk/openWB.log zu finden
Das sollte hier auch Aufschluss geben.

drego83, 2020-10-08:
und wie komme ich an die log ran, wenn ich eine fertige openWB gekauft habe? Ich habe keinen Zugriff auf die log ausgehend von der Webseite ...

snaptec, 2020-10-14:
Das geht auch aus dem Browser aufrufen @drego83

drego83, 2020-10-14:
Kannst Du mir bitte verraten wo und wie? Danke

snaptec, 2020-10-15:
http://ipderopenWB/openWB/ramdisk/openWB.log

Der-Kieler, 2020-10-19:
Es wäre toll, wenn das log (und vielleicht wesentliche weitere) über die Oberfläche erreichbar wären.
Würde viele Fragen vermeiden.

JanFellner, 2020-10-19:
Sehe ich aktuell genauso.

Speziell wünschen würde ich mir hierbei eine Seite in der ich die letzten 10/20/50 Log Einträge einsehen kann da das Öffnen der Datei ohne Console aktuell praktisch nicht nutzbar ist. Muss im Browser immer erst ganz nach unten Scrollen.
Die Ansicht könnte optional live laufen, sprich immer die letzten X Einträge anzeigen.

Weiterhin eine Seite in der ich immer live die letzte Berechnung sehe.

snaptec, 2020-10-26:
Unter Status ist das Debug Log nun auch sichtbar

closed 2020-10-26 by snaptec