snaptec/openWB #626: issues found in mqtt topics

benderl, 2020-10-02:
Ich sehe mir gerade die MQTT-Topics genauer an und bin über ein paar Dinge gestolpert.

Teilweise werden die set-Topics nach der Verarbeitung auf "0" oder einen anderen Wert (z. B. "empty") gesetzt. Ist das so richtig? Normalerweise wird das Topic komplett geleert.
https://github.com/snaptec/openWB/blob/75db3a989b3b7eb69706869e9d87065ded80d542/runs/mqttsub.py#L709
https://github.com/snaptec/openWB/blob/75db3a989b3b7eb69706869e9d87065ded80d542/runs/mqttsub.py#L623
https://github.com/snaptec/openWB/blob/75db3a989b3b7eb69706869e9d87065ded80d542/runs/mqttsub.py#L643
https://github.com/snaptec/openWB/blob/75db3a989b3b7eb69706869e9d87065ded80d542/runs/mqttsub.py#L678
https://github.com/snaptec/openWB/blob/75db3a989b3b7eb69706869e9d87065ded80d542/runs/mqttsub.py#L699
https://github.com/snaptec/openWB/blob/75db3a989b3b7eb69706869e9d87065ded80d542/runs/mqttsub.py#L679

Bei diesem Topic wird lediglich das set geleert, jedoch nichts weiter verarbeitet:
https://github.com/snaptec/openWB/blob/75db3a989b3b7eb69706869e9d87065ded80d542/runs/mqttsub.py#L474

An diesen Stellen wird fix Lp5 angesprochen, obwohl das Topic variabel für alle Lp aufgerufen wird und "devicenumb" zur Verfügung steht:
https://github.com/snaptec/openWB/blob/75db3a989b3b7eb69706869e9d87065ded80d542/runs/mqttsub.py#L250
https://github.com/snaptec/openWB/blob/75db3a989b3b7eb69706869e9d87065ded80d542/runs/mqttsub.py#L254

Sind das Bugs oder ist das alles so gewollt?

snaptec, 2020-10-02:
Die Graph Sachen werden aktiv befragt gefüllt und wieder auf empty gesetzt, heißt das stimmt so.

RequestDebug müsste wieder geleert werden

topicSender war/ist zur Analyse woher bzw. wer eine Änderung durchgeführt hat.
Das lokale standard/dark theme sendet hier die clientuid, web.openwb.de tut dies auch.
Ist dort etwas drinnen taucht es im MQTT.log auf Zeile 987 sorgt dafür es in das Log kommt, die Zeile 679 leert es schlicht dann nur.

Mit der 5 ist schlicht ein Bug - zeigt aber wie selten Lademenge bei Ladepunkt 3-8 genutzt wird.

benderl, 2020-10-02:
OK, schicke Dir in den nächsten Tagen einen PR.

closed 2020-10-20 by snaptec