snaptec/openWB #659: Support für VARTA pulse / pulse neo Speicher

amapparat, 2020-10-19:
VARTA hat die bisherige Schnittstelle in den neueren Speichern entfernt und erlaubt nur noch den Zugriff per modbus.
Ich konnte den Speicher mit einer eigenen php-modbus-Lösung auf auf einem Raspi in die openwb bringen, eine direkte Lösung wäre aber für die meisten Anwender sicher schöner.

Die relevanten Modbus-Adresses wären:
'int16', 'address' => 1066 => 'batterypower'
'uint16', 'address' => 1068 => 'soc'
'int16', 'address' => 1078 'gridpower'

Eine komplette Dokumentation zu den modbus Adressen hätte ich vorliegen.

Ansonsten vielen Dank für eure großartige openWB!

snaptec, 2020-10-20:
Wurde kürzlich erst implementiert - gerne in der Nightly nutzen.
siehe: https://github.com/snaptec/openWB/commit/a6a9d5d8beaa69784402bb3c526c121df40eea2e

closed 2020-10-20 by snaptec

ffpj, 2024-02-23:
Meine Konfig: Solaredge --1502-->Varta Pulse neo ---502?----->openWB series2  (Ver 1.9.303.0) ---??--->MyPV ACThor
habe die openWB hinter der Varta Puls neo: Modulkonfig Batterie und EVU klappt - die gelieferten Daten sind korrekt. ACThor = Ok.  Die openWB zeigt aktuell keine korrekten PV-Leistungsdaten an. = auch klar,  den in derModulkonfig PV ist keine Varta Eintrag möglich - wobei die Varta die PV-Daten des Solaredge verfügbar hat und selber auch anzeigt. Was muss in Modulkonfig PV eingestellt werden - um den korrekten PV  Ertrag auf der openWB sehen zu können?

Danke!!

benderl, 2024-02-24:
@ffpj Diskussionen sind besser im Forum aufgehoben.