snaptec/openWB #673: Negative Werte bei geladenen kW/km

JanFellner, 2020-10-31:
Feststellung: Stellt man beim sofort laden einen bestimmten kw/h Wert ein der geladen werden soll und wechselt anschließend (ohne das dieser Wert jemals erreicht wurde) in einen anderen Lade Modus stehen in der Übersicht negative Ladewerte.
Erwartung: Wechsle ich den Lademodus werden diese negativen Werte zurück gesetzt und der Zähler beginnt von 0
[image]

snaptec, 2020-10-31:
Um was für einen Ladepunkt handelt es sich?

JanFellner, 2020-11-01:
SimpleEVSEWifi mit aktuellster Firmware
Current Firmware Version: 1.0.2

OpenWb läuft mit
installierte Version: 1.8.412Nightly (2020-10-19 08:39:47 +0200 [104333f0] )

JanFellner, 2020-11-20:
Hab das Problem leider immer noch, praktisch jedes Mal wenn ich die openwb Webseite aufrufe. Neuste Version is.  @``

JanFellner, 2021-02-06:
Any news on this one? I still face it so I can never tell how much I charged after the charging has completed... 😔

benderl, 2021-02-06:
@JanFellner kannst Du bitte die genauen Schritte aufführen, um das Verhalten nachstellen zu können? Sind danach auch im Ladelog Einträge mit negativen Werten vorhanden?

JanFellner, 2021-02-06:
@benderl naja, ich stecke das Auto an :). Ich schaue in die Weboberfläche und sehe negative Werte.
Um sagen zu können wie es Auftritt müsste ich zunächst verstehen wann dieser Wert denn überhaupt gelöscht wird.

[image]

Heute früh beim Laden standen wieder negative Werte drinn.
Im Ladelog sehe ich nichts auffälliges.

Gibt es denn mittlerweile eine irgendwie geartete From von Diagnosedaten welche man an ein solches Ticket anhängen könnte?

Jan

JanFellner, 2021-02-18:
Heute erneut. Negative Werte in der UI.
Fragen meinerseits hierzu hatte ich ja schon geschrieben.

[image]
[image]

JanFellner, 2021-02-19:
Und wieder. Negative Werte... 
[image]

JanFellner, 2021-10-09:
Bald habe ich Jahrestag... 
[image]

Daher nochmals die Frage: Wann wird der Zähler zurückgesetzt/initialisiert?

JanFellner, 2021-10-30:
ich schließe dieses ticket da es auf rückfragen seitens der entwickler leider keinerlei feedback gibt.

closed 2021-10-30 by JanFellner