snaptec/openWB #674: Längenbegrenzung bei 1. Smart Home Gerät

JanFellner, 2020-10-31:
Der Name des 1. Smart home Gerätes scheint beim Speichern/Laden irgendwo verkürzt zu werden. Ich würde gerne Waschmaschine eintragen. Die Ui erlaubt dies. Der Name wird aber nicht übernommen. Beim 2. Smart home gerät tritt dieses Problem so nicht auf.
[image]

snaptec, 2020-10-31:
Was steht im MQTT Log (unter Status einsehbar)?
Bis wieviele Zeichen geht?

JanFellner, 2020-11-01:
[image]

Im SmartHomeLog steht auch nur Waschm.

Egal was ich eintrage und speichere, es steht am Scluss wieder auf Waschm.

closed 2020-12-04 by snaptec

JanFellner, 2020-12-04:
closed fixed?
closed cannot reproduce?
closed won´t fix?

benderl, 2020-12-04:
Fixed: #760