snaptec/openWB #675: Smarthome 2.0 Problem Tasmota ohne Leistungsmessung

StefanKi, 2020-10-31:
Ich habe eine Tasmota Steckdose ohne Leistungsmessung.
Diese funktioniert mit der Version 1.8.400Beta nicht.
Das Problem ist, dass bei mir (ich vermute, dass das wegen der fehlenden Leistungsmessung ist), der Status 8 keinen Inhalt außer Uhrzeit und Datum liefert:
`{"StatusSNS":{"Time":"2020-10-31T13:37:02"}}`

Ich habe den Code folgendermaßen adaptiert (ab Zeile 345):
`                   answer = json.loads(str(urllib.request.urlopen("http://"+config.get('smarthomedevices', 'device_ip_'+str(numberOfDevices))+"/cm?cmnd=Status", timeout=3).read().decode("utf-8")))
                    watt=0
                    if ( int(answer['Status']['Power']) > 0 ):
                        relais=1
                    else:
                        relais=0
`
Damit funktioniert es dann. (Ist natürlich nicht die fertige Lösung, aber der Test, dass es dann geht, weil ja sonst die Watt falsch sind)

benderl, 2021-01-12:
Bitte nochmal mit dem überarbeiteten SmartHome 2.0 testen. Da hat sich seit Oktober viel verändert.

benderl, 2021-02-13:
Anscheinend besteht kein weiteres Interesse.

closed 2021-02-13 by benderl