snaptec/openWB #684: Zurücksetzen der Ladestatistiken

tomac01, 2020-11-08:
Hallo,

als Benutzer würde ich gerne meine Ladestatistiken über die Einstellungen in der Weboberfläche zurücksetzen können.

Hintergrund:
Insbesondere am Anfang bei der Konfiguration habe ich mehrere Werte und Moduleinstellungen durchprobiert. Da nun alle Einstellungen feststehen und alles läuft, würde ich gerne einen definierten Aufzeichungsanfang haben und dafür alle bisherigen Aufzeichnungen löschen wollen.

Grüße,
tomac

snaptec, 2020-11-08:
Nur das Ladelog oder die Graph Daten?

tomac01, 2020-11-08:
Beides als getrennte Optionen wäre wohl am besten.

snaptec, 2020-11-09:
In meinen Augen müsste speziell das Ladelog aber noch in irgendeiner Form "gesichert" werden. Oft ist das für FA oder AG relevant. Ein "einfaches" löschen halte ich da nicht für sinnvoll.
Meinungen?

tomac01, 2020-11-09:
Mhm, das Löschen würde ich nicht mehr machen, wenn alles läuft; also im Regelbetrieb. Mein Anforderungsidee war eine Art Zurücksetzen der Zähler, um „richtig“ starten zu können. Die Daten bis dahin werden in diesem Fall nicht benötigt; ein Backup dementsprechend auch nicht.

Eine andere Idee (neue Anforderung?) wäre zusätzlich der Export und Import von Ladedaten per CSV oder XML.

Damit könnte Mann ggf. Alternativ zum Löschen, die Daten anpassen.

snaptec, 2020-11-09:
Der Export des Ladelog per CSV ist bereits implementiert. der Import nicht. 

Speziell die Graph Daten basieren auf den absoluten Zählerständen, da sollte eigentlich nur der Tag des einrichtens Krumm sein, danach sollte es passen.

toemk1, 2020-11-20:
Ich bin nicht sicher, ob das hier passt, aber bei mir fehlt durch einen Abstutz der OpenWB (ausgelöst durch Auslösen des FIs) ein Ladevorgang im Ladelog. Eine manuelle Anpassung oder ein Abgleich mit dem Zähler wäre also aus meiner Sicht hilfreich.