snaptec/openWB #695: EVU Zähler via MQTT lückenhaft

mode2k, 2020-11-10:
Ich sende Zählerstände und Verbrauchsdaten via MQTT an openWB:
 **mqtt.log**
`2020-11-10 18:06:17 Topic: openWB/set/evu/WhImported Message: 27798741.900000002
2020-11-10 18:06:17 Topic: openWB/set/evu/WhExported Message: 22975364.4
2020-11-10 18:06:17 Topic: openWB/set/evu/W Message: 616.5
2020-11-10 18:06:19 Topic: openWB/set/evu/WhImported Message: 27798742.2
2020-11-10 18:06:19 Topic: openWB/set/evu/WhExported Message: 22975364.4
2020-11-10 18:06:19 Topic: openWB/set/evu/W Message: 597.9
2020-11-10 18:06:21 Topic: openWB/set/evu/WhImported Message: 27798742.5
2020-11-10 18:06:21 Topic: openWB/set/evu/WhExported Message: 22975364.4
2020-11-10 18:06:21 Topic: openWB/set/evu/W Message: 595.8
2020-11-10 18:06:22 Topic: openWB/set/evu/WhImported Message: 27798742.8
2020-11-10 18:06:22 Topic: openWB/set/evu/WhExported Message: 22975364.4
2020-11-10 18:06:22 Topic: openWB/set/evu/W Message: 610.9
2020-11-10 18:06:24 Topic: openWB/set/evu/WhImported Message: 27798743.1
2020-11-10 18:06:24 Topic: openWB/set/evu/WhExported Message: 22975364.4
2020-11-10 18:06:24 Topic: openWB/set/evu/W Message: 623.2
2020-11-10 18:06:26 Topic: openWB/set/evu/WhImported Message: 27798743.4
2020-11-10 18:06:26 Topic: openWB/set/evu/WhExported Message: 22975364.4
2020-11-10 18:06:26 Topic: openWB/set/evu/W Message: 620.9
`
Dabei seit der Graph sehr lückenhaft aus.
[image]

Anbei noch ein Blick auf die Messages vom Mosquitto:
`openWB/graph/3alllivevalues 18:04:17,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:04:27,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:04:37,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:04:47,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:04:57,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:05:07,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:05:17,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:05:27,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:05:37,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:05:47,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:05:57,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:06:06,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:06:16,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:06:26,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:06:36,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:06:46,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:06:56,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:07:07,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:07:17,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:07:27,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:07:37,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:07:47,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:07:57,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:08:06,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:08:16,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:08:26,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:08:36,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:08:46,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:08:56,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:09:07,0,0,0,0,0,0,,,0,,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
18:09:17,638,0,0,0,0,0,,,0,,638,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
openWB/graph/16alllivevalues -
openWB/graph/8alllivevalues -
openWB/graph/11alllivevalues -
openWB/graph/9alllivevalues -
openWB/graph/7alllivevalues -
openWB/graph/10alllivevalues -
openWB/graph/12alllivevalues -
openWB/graph/13alllivevalues -
openWB/graph/14alllivevalues -
openWB/graph/15alllivevalues -
openWB/graph/5alllivevalues -
openWB/global/strLastmanagementActive
openWB/evu/WhImported 27798773.099999998
openWB/set/evu/WhImported 27798773.4
openWB/set/evu/WhExported 22975364.4
openWB/set/evu/W 609.6
openWB/set/evu/WhImported 27798773.700000003
openWB/set/evu/WhExported 22975364.4
openWB/set/evu/W 612.2
openWB/set/evu/WhImported 27798774
openWB/set/evu/WhExported 22975364.4
openWB/set/evu/W 643.5
openWB/set/evu/WhImported 27798774.3
openWB/set/evu/WhExported 22975364.4
openWB/set/evu/W 606.6
`

Ich habe die EVU Glättung mit 10 Sekunden aktiviert, das hat aber nicht verbessert.

In /var/log/openWB.log gibt es keine relevanten Ausgaben.

snaptec, 2020-11-10:
Watt als mit übergeben, nicht als float, stimmt es dann?

mode2k, 2020-11-10:
Super, das was es. Mit Int Wert ohne Nachkomma funktioniert es.
-->Solved. 
Vielen Dank

closed 2020-11-10 by mode2k