snaptec/openWB #732: Was tun bei mehreren Wechselrichtern?

AntonDerZauberer, 2020-11-25:
Hi Andi, 
ich bin absolut neu im Thema Ladesteuerung. Ich würde gerne meinen bald kommenden Zoe mit überschüssigem Strom laden. Dafür würde ich gerne deine Lösung nutzen, da sie einfacher einzurichten sein soll, als OpenWb. Ich würde gerne eine Wallbe Eco 2.0s nutzen. Lässt sich die Box richtig steuern, oder geht sie nur an und aus, wenn eine bestimmte Grenze überschritten wird? Ich habe drei zehn Jahre alte SMA Wechselrichter ohne smarter Technik und weit entfernt von der Wallbox. Muss ich dann für jeden Wechselrichter einen eigenen Modbusadapter kaufen, oder würde es Sinn ergeben, dass z.B. ein Fronius Smartmeter nutzen würde? Lässt sich das einbinden?

Ich freue mich auf mein Auto und besonders auf deine Antwort :))))

MfG Anton

snaptec, 2020-11-26:
Sicher das du hier richtig bist?
Dein Setup geht mit openWB, nicht aber mit Wallbe. Das Setup ist per EVU Kit oder auch Fronius Smartmeter in der openWB per Webinterface in 3 Minuten konfiguriert.

AntonDerZauberer, 2020-11-26:
Oh, tut mir leid. Ich wollte diese Nachricht an die "Konkurrenz" schicken :). Ich wünsche noch viel Erfolg bei diesem Projekt.

closed 2020-11-28 by snaptec